genodelabs/genode #3236: wm: ep deadlocks during destroy_view handling

alex-ab, 2019-03-18:
At

https://github.com/genodelabs/genode/blob/master/repos/gems/src/server/wm/nitpicker.h#L881

a locked pointer is taken. During the destruction of the view for the release is waited for at

https://github.com/genodelabs/genode/blob/master/repos/gems/src/server/wm/nitpicker.h#L269

which never happens.

alex-ab, 2019-03-18:
@nfeske: you are more knowledgeable than me regarding the wm, do you have an idea how to fix that ?

nfeske, 2019-03-25:
I have no good idea how to solve it in a non-quirky way without enhancing the `Weak_ptr` mechanism to support the destruction of weak objects.

As interim solution, I see two options. First, given that the wm is not multi-threaded, the "locked" part of the `locked_ptr` is actually not needed here. So we might drop the call of `lock_for_destruction`, or better replace the call by a comment that points to this issue. The second option would be to leak the raw pointer from the scope of the locked pointer and call destroy on the raw pointer beyond the scope of the locked pointer.

Unless, the first option produces ugly warning messages, I'd probably go for the first one.

nfeske, 2021-02-07:
Fixed by https://github.com/genodelabs/genode/commit/4908e4334380b8b60dca85b1219c0b8bf5cfbbb5

commit 4908e433 (2021-02-07): wm: fix deadlock in destroy-view handling

The deadlock reported in #3236 could be reproduced via the wm.run script
and the modification of test/nitpicker in commit "nitpicker: fix destroy
with invalid handle" by clicking on the testnit entry of the launchpad.

This patch fixes the deadlock by releasing the locked pointer early in
the destruction path, which is legitimate as the wm is single-threaded.

Fixes #3236
closed 2021-02-23 by nfeske