genodelabs/genode #3237: nic_router: manipulate port-forwarding port

m-stein, 2019-03-18:
It should be possible to change the destination port of requests that get port-forwarded by the NIC router. This was motivated by some use cases of @ssumpf and @cnuke. One hard reason to add the feature is the situation when we want client requests with different destination ports be handled by one and the same server.

m-stein, 2019-03-18:
Fixed:
bab5db589f nic_router: manipulate port-forwarding port

nfeske, 2019-03-18:
Cool! This is a very nice feature. Thanks @m-stein for adding it.

But does the new feature really require changing existing configuration options? I think that just adding the `to_port` attribute would be sensible and logical (because it adds further precision of the existing `to` attribute). The rename of the `to` attribute to `to_ip`, however, looks overzealous to me. It breaks all nic-router configurations that use IP forwarding today for no good reason.

m-stein, 2019-03-18:
It's a pure question of style, I like a descriptive interface more than one time convenience, but it's your decision.

nfeske, 2019-03-19:
Adjusting styles or configuration formats is of course sometimes desired but please consider that such a change is always a pain for users. For this reason, the adaptation to such breaking changes must be planned, prepared, justified, communicated, and carried out. These are costs that should not be produced lightheartedly. Remember that you are *not* the one who pays those costs, but the users of your software are.

Your off-handed remark that this is just a question of style or a one-time inconvenience completely misses this point.

For the record, my repulse against the renaming of `to` -> `to_ip` is not about my preference of style (I just wanted to express that I find the addition of an optional `to_port` attribute completely acceptable). It is about the breakage that you are willing to introduce as a mere side effect of adding a minor feature.

m-stein, 2019-03-19:
Adapted to the suggestions of @chelmuth and @nfeske:
467515960d Squash "nic_router: manipulate port-forwarding port" (revert to_ip, new commit message)
73365b2eed Fixup: "nic_router: manipulate port-forwarding port" (README port range)
7948af81cd nic_router: manipulate port-forwarding port

commit 2b73ceb2 (2019-03-19): nic_router: manipulate port-forwarding port

The new configuration attributes <tcp-forward to_port="123" /> and
<udp-forward to_port="123" /> enable manipulating the destination port of
port-forwarded packets.

Fixes #3237
commit 789d0db9 (2019-03-20): nic_router: manipulate port-forwarding port

The new configuration attributes <tcp-forward to_port="123" /> and
<udp-forward to_port="123" /> enable manipulating the destination port of
port-forwarded packets.

Fixes #3237
commit bcc049ce (2019-04-01): nic_router: manipulate port-forwarding port

The new configuration attributes <tcp-forward to_port="123" /> and
<udp-forward to_port="123" /> enable manipulating the destination port of
port-forwarded packets.

Fixes #3237
closed 2019-04-02 by nfeske