genodelabs/genode #3238: depot_autopilot: still some sporadic false-positives

skalk, 2019-03-19:
Since the integration of the `depot_autopilot` into the nightly test routines, there are sporadic false-positive errors. As an example, tonight the `depot_autopilot` reported that the `test-magic_ring_buffer` failed due to a 10 sec timeout on imx6q_sabrelite on the Hw kernel, although it printed the good-case message immediately within the first second.
This is just an example we could observe these failing of actually succeeding test regularily.

m-stein, 2019-03-19:
My last state regarding this topic is that I was not able to reproduce it. It was the reason we added the TEST_REPEAT variable to the depot_autopilot.run script. But doing tests in a loop didn't trigger it neither.

skalk, 2019-03-20:
@m-stein thanks for documenting the last approach. Nevertheless, we have to get rid of the false messages. If you cannot reproduce it, my suggestion would be to have some non-invasive interims changes on staging to identify the actual cause. It always happens on timeouts of a test. Maybe there is a way to check the output again afterwards using different mechanisms to detect the error case?

m-stein, 2019-03-20:
As a first step, we could write the log output of each test to a RAM buffer. If the test fails, we write the buffer content as hex values to to the output. This way, we could detect abnormal non-printable characters. As a second step, we could instead write the parsing steps of the Depot Autopilot to the buffer. I.e. the path taken for each character in the matching loop. Reading this is annoying work but it should definitely reveal the problem. I'll come up with appropriate patches.

skalk, 2019-03-21:
That sounds good, as long as it is printed in the error case only. Interestingly, tonight it was again `test-magic_ring_buffer` this time on top of Fiasco.OC/PBXA9 on Qemu.

m-stein, 2019-03-21:
These are some patches I'd like to suggest for staging only to be able to debug the nightly tests:
a5e461e3a7 NOT FOR MASTER "depot_autopilot.run: raise depot autopilot quota"
3b51b3e0ff NOT FOR MASTER "depot_autopilot: list log matches"
4855b19f16 NOT FOR MASTER "depot_autopilot: hex dump of failed test log"

They let the depot autopilot print some extra debug output for failed tests. First, a hex dump of the tests log is printed as can be seen in this example after the test has finished: 
```
[init -> depot_autopilot] --- Run "test-log" (max 5 sec) ---
[init -> depot_autopilot] 
[init -> depot_autopilot] 2.442 [init -> test-log] hex range:          [0e00,1680)
[init -> depot_autopilot] 2.491 [init -> test-log] empty hex range:    [0abc0000,0abc0000) (empty!)
[init -> depot_autopilot] 2.520 [init -> test-log] hex range to limit: [f8,ff]
[init -> depot_autopilot] 2.527 [init -> test-log] invalid hex range:  [f8,08) (overflow!)
[init -> depot_autopilot] 2.542 [init -> test-log] negative hex char:  0xfe
[init -> depot_autopilot] 2.566 [init -> test-log] positive hex char:  0x02
[init -> depot_autopilot] 2.572 [init -> test-log] floating point:     1.70
[init -> depot_autopilot] 2.578 [init -> test-log] multiarg string:    "parent -> child.7"
[init -> depot_autopilot] 2.596 [init -> test-log] String(Hex(3)):     0x3
[init -> depot_autopilot] 2.627 [init -> test-log] Very long messages:
[init -> depot_autopilot] 2.770 [init -> test-log -> log] 1......................................................................................................................................................................................................
[init -> depot_autopilot] ...............................2
[init -> depot_autopilot] 2.819 [init -> test-log] 3.............................................................................................................................................................................................................
[init -> depot_autopilot] ........................4
[init -> depot_autopilot] 2.849 [init -> test-log] 5.............................................................................................................................................................................................................
[init -> depot_autopilot] ........................6
[init -> depot_autopilot] 2.893 [init -> test-log] 
[init -> depot_autopilot] 2.899 [init -> test-log] Test done.
[init -> depot_autopilot] 
[init -> depot_autopilot] ------- captured log ---------
[init -> depot_autopilot] 00000000:  5b 69 6e 69  74 20 2d 3e  20 74 65 73  74 2d 6c 6f
[init -> depot_autopilot] 00000010:  67 5d 20 68  65 78 20 72  61 6e 67 65  3a 20 20 20
[init -> depot_autopilot] 00000020:  20 20 20 20  20 20 20 5b  30 65 30 30  2c 31 36 38
[init -> depot_autopilot] 00000030:  30 29 1b 5b  30 6d 0a 5b  69 6e 69 74  20 2d 3e 20
[init -> depot_autopilot] 00000040:  74 65 73 74  2d 6c 6f 67  5d 20 65 6d  70 74 79 20
[init -> depot_autopilot] 00000050:  68 65 78 20  72 61 6e 67  65 3a 20 20  20 20 5b 30
[init -> depot_autopilot] 00000060:  61 62 63 30  30 30 30 2c  30 61 62 63  30 30 30 30
[init -> depot_autopilot] 00000070:  29 20 28 65  6d 70 74 79  21 29 1b 5b  30 6d 0a 5b
[init -> depot_autopilot] 00000080:  69 6e 69 74  20 2d 3e 20  74 65 73 74  2d 6c 6f 67
[init -> depot_autopilot] 00000090:  5d 20 68 65  78 20 72 61  6e 67 65 20  74 6f 20 6c
[init -> depot_autopilot] 000000a0:  69 6d 69 74  3a 20 5b 66  38 2c 66 66  5d 1b 5b 30
[init -> depot_autopilot] 000000b0:  6d 0a 5b 69  6e 69 74 20  2d 3e 20 74  65 73 74 2d
[init -> depot_autopilot] 000000c0:  6c 6f 67 5d  20 69 6e 76  61 6c 69 64  20 68 65 78
[init -> depot_autopilot] 000000d0:  20 72 61 6e  67 65 3a 20  20 5b 66 38  2c 30 38 29
[init -> depot_autopilot] 000000e0:  20 28 6f 76  65 72 66 6c  6f 77 21 29  1b 5b 30 6d
[init -> depot_autopilot] 000000f0:  0a 5b 69 6e  69 74 20 2d  3e 20 74 65  73 74 2d 6c
[init -> depot_autopilot] 00000100:  6f 67 5d 20  6e 65 67 61  74 69 76 65  20 68 65 78
[init -> depot_autopilot] 00000110:  20 63 68 61  72 3a 20 20  30 78 66 65  1b 5b 30 6d
[init -> depot_autopilot] 00000120:  0a 5b 69 6e  69 74 20 2d  3e 20 74 65  73 74 2d 6c
[init -> depot_autopilot] 00000130:  6f 67 5d 20  70 6f 73 69  74 69 76 65  20 68 65 78
[init -> depot_autopilot] 00000140:  20 63 68 61  72 3a 20 20  30 78 30 32  1b 5b 30 6d
[init -> depot_autopilot] 00000150:  0a 5b 69 6e  69 74 20 2d  3e 20 74 65  73 74 2d 6c
[init -> depot_autopilot] 00000160:  6f 67 5d 20  66 6c 6f 61  74 69 6e 67  20 70 6f 69
[init -> depot_autopilot] 00000170:  6e 74 3a 20  20 20 20 20  31 2e 37 30  1b 5b 30 6d
[init -> depot_autopilot] 00000180:  0a 5b 69 6e  69 74 20 2d  3e 20 74 65  73 74 2d 6c
[init -> depot_autopilot] 00000190:  6f 67 5d 20  6d 75 6c 74  69 61 72 67  20 73 74 72
[init -> depot_autopilot] 000001a0:  69 6e 67 3a  20 20 20 20  22 70 61 72  65 6e 74 20
[init -> depot_autopilot] 000001b0:  2d 3e 20 63  68 69 6c 64  2e 37 22 1b  5b 30 6d 0a
[init -> depot_autopilot] 000001c0:  5b 69 6e 69  74 20 2d 3e  20 74 65 73  74 2d 6c 6f
[init -> depot_autopilot] 000001d0:  67 5d 20 53  74 72 69 6e  67 28 48 65  78 28 33 29
[init -> depot_autopilot] 000001e0:  29 3a 20 20  20 20 20 30  78 33 1b 5b  30 6d 0a 5b
[init -> depot_autopilot] 000001f0:  69 6e 69 74  20 2d 3e 20  74 65 73 74  2d 6c 6f 67
[init -> depot_autopilot] 00000200:  5d 20 56 65  72 79 20 6c  6f 6e 67 20  6d 65 73 73
[... cut ...]
[init -> depot_autopilot] [0000050f..00000526]
[init -> depot_autopilot] [00000415..00000427]
[init -> depot_autopilot] [0000031b..0000032d]
[init -> depot_autopilot] [00000215..0000022a]
[init -> depot_autopilot] [00000000..00000204]
[init -> depot_autopilot] ------------------------------
```

Together, this can tell us where the mismatch occurs in the sporadicly failing tests without having to reproduce it. The commits are written in a way that they should affect the nightly tests as less as possible.

skalk, 2019-03-22:
@m-stein that sounds like a really good measure to me. I'm curious whether it helps identifying the problem. Thanks for providing it!

nfeske, 2019-03-22:
The fs_tool test still sporadically fails for another reason. The order of the log lines of the tested components not completely deterministic. The test wants to match the output of two components involved in the test, reports generated by an fs_query instance and the actual fs_tool. I have to think of a way about how to make them sequential.

m-stein, 2019-03-22:
If the feature proves useful in the context of this issue, I would suggest to add it to the normal feature set of the depot autopilot. It could serve well when having to write log events for a new test and they refuse to work as expected. This happened a few times to me and although it wasn't caused by any obscurity in the parsing mechanism but simply by my malicious log pattern, it wasn't easy to debug without any tooling. I'd let the feature be disabled by default, so, it must be enabled in the configuration.

m-stein, 2019-03-22:
@nfeske I sense, that this is a very common problem in our tests and I thought about solving it in the depot autopilot by providing the means for saying "render a test succeeded when pattern A _and_ pattern B _and_ pattern C ... are fullfilled". Something like:
```
<events>
   <log meaning="succeeded">
      <and>child2 finished</and>
      <and>child1 finished</and>
   </log>
</events>
```
What do you think about this?

m-stein, 2019-03-22:
I'd like to add that, AFAICS, this wouldn't be hard to implement. Actually it wouldn't change much in the current implementation.

nfeske, 2019-03-22:
That looks like a useful feature. But it overloads the meaning of the content of the <log> node. Normally, the content of this node is the log output. Now, with the <and> node, there would be a special case. Another question is what would happen if somebody likes to have an <or> condition later? Should we anticipate that somehow?

To fix the CI issue in the short term, I will rework the test nevertheless. So the depot_autopilot improvements are not on the critical path.

m-stein, 2019-03-25:
@nfeske What do you mean "to have an condition later" ? If you mean "match one pattern first and the other one afterwards" this would be done by using the * wildcard. Regarding the overloading, I agree that this is a bit ugly. How about this instead:
```
<events>
   <log meaning="succeeded">
      Child1 finished
      <and/>
      Child2 finished
   </log>
</events>
```
It would not overload but enhance the meaning of the <log> content. Also, it would not render existing runtime files invalid.
I also don't see this issue as critical.

nfeske, 2019-03-25:
> I have to think of a way about how to make them sequential.

Commit https://github.com/genodelabs/genode/commit/e04668eefb752c6daf9b267f67a239851a039198 changes the test-fs_tool to make its output deterministic.

nfeske, 2019-03-25:
> ... this is a bit ugly.

@m-stein I think it is a bit unfortunate that you disregarded my original suggestion when we discussed the way of how the depot-autopilot should detect failure/success. Back then, you just went for your "events" declaration and apparently dismissed my proposal as just a matter of taste. To recap, my suggestion was in the line of the `expect_init_state` mechanism of the init test. Applied to the depot autopilot, this approach would have looked likes this:

```
<expect>
  <not> <timeout seconds="30"/> </not>
  <log>one</log>
  <log>another</log>
</expect>
```

The `<expect>` node is a plain condition for success. The condition is fulfilled if all elements are true. The terms `event`, or `meaning` are not needed.

For the common case where a timeout indicates failure, a shortcut for `<not> <timeout seconds="..."/> </not>` could be added in the form of a `timeout_seconds="<N>"` attribute to the expect node. So the above example would become:

```
<expect timeout_seconds="30">
  <log>one</log>
  <log>another</log>
</expect>
```

The (rare) case where a timeout is actually the good case and expected would be covered by

```
<expect>
   <timeout seconds="30"/>
   <not> <log>this should never happen</log> </not>
</expect>
```

When contrasting this clarity with your suggestion above, I think that the route you took was appropriate to reach the current state (which is already very cool!). But the direction should not be traveled much farther. As you just said, it becomes not more pretty. If we intend to make the runtime declarations more expressive, we should reconsider the `<events>` approach in the first place.

m-stein, 2019-03-25:
Sorry, I definitely should have communicated this better. My intention was not to disregard any other approach. I'm not sure about which one is "better" in the end and just wanted to start with one.

Actually, the thought that I had when using the <events> approach over the <expect> approach was not that it's just a matter of taste. The <expect> approach depends on the fact that there are only two actions an event can trigger, succeed and fail. I thought it would be nice to keep the new tool open as a mere event-handling mechanism as this might be something you want to do also in other contexts than the depot autopilot with individual actions triggered by events.

I might be totally wrong with this and I have no problem at all to switch to the <expect> approach but I don't share your feeling that they differ a lot when it comes to clarity or beauty. With <expect> we silently apply an AND operator to all events and relate a fulfilled <expect> to "test succeeded" respectively a missed <expect> to "test failed".

nfeske, 2019-04-02:
Yesterday, I temporarily removed the instrumentation from staging in order to update the master branch today. Interestingly, this night's results (w/o the instrumentation) seem to contain quite a few more false negatives compared to the days before. Heisenberg effect?

I just re-added the instrumentation to staging again.

m-stein, 2019-04-02:
@nfeske Thanks for doing so. Would be reasonable that the diagnostics influence the bug. They definitely slow down the log parsing a bit. I'll reserve some time for it today.

m-stein, 2019-04-02:
I didn't find the time to investigate today because of discussions with the others. I'll try to go on tomorrow.

rite, 2019-06-17:
I'm not sure if it is related, but I encounter sporadic failures as well. To get some diagnosics I let the `depot_autopilot` repeat `until_failed` on x86_64/linux with the `NOT FOR MASTER` commits. In the second iteration, `test-xml_generator` failed, printing the output below. Let me know if you'd like to have the complete log or this behavior is not related to this issue.

```
...
[13:26:23.041] [init -> depot_autopilot] 2.672 [init -> test-weak_ptr] thread: destruction completed, job done
[13:26:23.041] [init -> depot_autopilot] 2.673 [init -> test-weak_ptr] 
[13:26:23.041] [init -> depot_autopilot] 2.673 [init -> test-weak_ptr] --- finished test-weak_ptr ---
[13:26:23.041] [init -> depot_autopilot] 
[13:26:23.042] [init -> depot_autopilot]  test-weak_ptr                   ok         2.673  log "--- finished test-weak_ptr ---"
[13:26:23.065] [init -> depot_autopilot] 
[13:26:23.065] [init -> depot_autopilot] --- Run "test-xml_generator" (max 20 sec) ---
[13:26:23.066] [init -> depot_autopilot] 
[13:26:23.116] [init -> depot_autopilot] 0.049 [init -> test-xml_generator] --- XML generator test started ---
[13:26:23.120] [init -> vfs] session opened for 'dynamic -> test-xml_generator -> test-xml_generator -> gcov_data' at '/gcov_data'
[13:26:23.121] [init -> depot_autopilot] 0.055 [init -> test-xml_generator] 
[13:26:23.121] [init -> depot_autopilot] 0.055 [init -> test-xml_generator] used 308 bytes, result:
[13:26:23.121] [init -> depot_autopilot] 0.055 [init -> test-xml_generator] 
[13:26:23.121] [init -> depot_autopilot] 0.055 [init -> test-xml_generator] <config xpos="27" ypos="34">
[13:26:23.122] [init -> depot_autopilot] 0.056 [init -> test-xml_generator]     <box width="320" height="240"/>
[13:26:23.122] [init -> depot_autopilot] 0.056 [init -> test-xml_generator]     <label name="a test">
[13:26:23.122] [init -> depot_autopilot] 0.056 [init -> test-xml_generator]             <sub_label/>
[13:26:23.123] [init -> depot_autopilot] 0.057 [init -> test-xml_generator]             <another_sub_label>
[13:26:23.123] [init -> depot_autopilot] 0.057 [init -> test-xml_generator]                     <sub_sub_label/>
[13:26:23.123] [init -> depot_autopilot] 0.057 [init -> test-xml_generator]             </another_sub_label>
[13:26:23.123] [init -> depot_autopilot] 0.057 [init -> test-xml_generator]     </label>
[13:26:23.124] [init -> depot_autopilot] 0.058 [init -> test-xml_generator]     <bool true="true" false="false"/>
[13:26:23.124] [init -> depot_autopilot] 0.058 [init -> test-xml_generator]     <signed int="-1" long="-2" longlong="-3"/>
[13:26:23.124] [init -> depot_autopilot] 0.058 [init -> test-xml_generator]     <unsigned int="1" long="2" longlong="3"/>
[13:26:23.125] [init -> depot_autopilot] 0.059 [init -> test-xml_generator] </config>
[13:26:23.125] [init -> depot_autopilot] 0.059 [init -> test-xml_generator] 
[13:26:23.125] [init -> depot_autopilot] 0.059 [init -> test-xml_generator] buffer exceeded (expected error)
[13:26:23.126] [init -> depot_autopilot] 0.060 [init -> test-xml_generator] 
[13:26:23.126] [init -> depot_autopilot] 0.060 [init -> test-xml_generator] exception with value 10 on level 4 (expected error)
[13:26:23.126] [init -> depot_autopilot] 0.060 [init -> test-xml_generator] exception with value 20 on level 8 (expected error)
[13:26:23.127] [init -> depot_autopilot] 0.061 [init -> test-xml_generator] exception with value 30 on level 4 (expected error)
[13:26:23.127] [init -> depot_autopilot] 0.061 [init -> test-xml_generator] exception with value 11 on level 4 (expected error)
[13:26:23.127] [init -> depot_autopilot] 0.061 [init -> test-xml_generator] exception with value 21 on level 8 (expected error)
[13:26:23.128] [init -> depot_autopilot] 0.062 [init -> test-xml_generator] exception with value 31 on level 4 (expected error)
[13:26:23.128] [init -> depot_autopilot] 0.062 [init -> test-xml_generator] exception with value 12 on level 4 (expected error)
[13:26:23.128] [init -> depot_autopilot] 0.062 [init -> test-xml_generator] exception with value 22 on level 8 (expected error)
[13:26:23.129] [init -> depot_autopilot] 0.063 [init -> test-xml_generator] exception with value 32 on level 4 (expected error)
[13:26:23.129] [init -> depot_autopilot] 0.063 [init -> test-xml_generator] exception with value 40 on level 2 (expected error)
[13:26:23.130] [init -> depot_autopilot] 0.063 [init -> test-xml_generator] 
[13:26:23.130] [init -> depot_autopilot] 0.064 [init -> test-xml_generator] used 419 bytes, result:
[13:26:23.130] [init -> depot_autopilot] 0.064 [init -> test-xml_generator] 
[13:26:23.130] [init -> depot_autopilot] 0.064 [init -> test-xml_generator] <config>
[13:26:23.130] [init -> depot_autopilot] 0.064 [init -> test-xml_generator]     <level1>
[13:26:23.131] [init -> depot_autopilot] 0.065 [init -> test-xml_generator]             <level2 attr1="2271560481">
[13:26:23.131] [init -> depot_autopilot] 0.065 [init -> test-xml_generator]                     <level3 attr1="Hallo" attr2="123000">
[13:26:23.131] [init -> depot_autopilot] 0.065 [init -> test-xml_generator]                             <level4_1 attr1="true" attr2="Welt"/>
[13:26:23.132] [init -> depot_autopilot] 0.066 [init -> test-xml_generator]                             <level4_2/>
[13:26:23.132] [init -> depot_autopilot] 0.066 [init -> test-xml_generator]                     </level3>
[... cut ...]
[13:26:23.181] [init -> depot_autopilot] 
[13:26:23.181] [init -> depot_autopilot]  test-xml_generator              failed     0.070  log "Error: "
[13:26:23.200] [init -> depot_autopilot] 
[13:26:23.200] [init -> depot_autopilot] --- Run "test-xml_node" (max 20 sec) ---
[13:26:23.201] [init -> depot_autopilot] 
[13:26:23.232] [init -> depot_autopilot] 0.031 [init -> test-xml_node] --- XML-token test ---
[13:26:23.233] [init -> depot_autopilot] 0.032 [init -> test-xml_node] token type="SINGLECHAR", len=1, content="<"
...
```

m-stein, 2019-06-17:
@rite Thanks for documenting this error so well!
I would say that this is not a false positive as the depot autopilot correctly complains about  an error at the end of the test:
```
[init -> test-xml_node] --- End of XML-parser test ---
 [init -> depot_autopilot] 0.070 [init -> test-xml_generator] Error: Uncaught exception of type 'File_system::Node_already_exists'
```
This would be expected instead:
```
[init -> test-xml_node] --- End of XML-parser test ---
[init] child "test-xml_node" exited with exit value 0
```
But anyway, it seems to be a bug in the XML generator test and should be fixed. I'll try to reproduce it.

m-stein, 2019-06-17:
Sorry, I had copied from the wrong package runtime and had to fix this in my previous comment.

rite, 2019-06-18:
Indeed it isn't a false positive. It looks like `test-xml_generator` fails reproducibly in the second iteration.
I removed the test package and again let `run/depot_autopilot` run until `until_failed`. 8 iterations run successfully - in the 9th iteration two tests failed:

- **test-fault_detection:** no false positive, it times out because of an unhandled resource request  (see [1] line ~45985).

- **test-fs_rom_update:** no false positive as well see ([1] line ~46117).

This errors came up while debugging sporadic pipeline failures but don't quite fit to this issue. Shall I create a separate issue to keep track? One for each test package? Any other suggestion? @m-stein please let me know what you prefer.

-- 
[1] https://gist.github.com/rite/5f5105b899c176fe982971027a52b780

m-stein, 2019-06-18:
@rite The problem with test-fs_rom_update looks like it is related to issue https://github.com/genodelabs/genode/issues/3398 (already fixed). The problem with test-fault_detection seems to be a memory leak in the depot_autopilot. I assume it isn't related to test-fault_detection nor to your pipeline failures. Anyway, it should receive a dedicated issue. Thanks again for documenting!

chelmuth, 2019-12-20:
There were no strange false positives since the log mechanism developments.

closed 2019-12-20 by chelmuth