genodelabs/genode #3239: Native CMake

ehmry, 2019-03-19:
This is probably going to be a long term issue.

The Qt projects that I have looked at for port using the SDK all use CMake, so it makes sense that a CMake port should be attempted before re-implementing the build systems of these projects. Using CMake to cross-compile is an option, but I would prefer to attempt this natively so not to mix the host and target systems. This is probably not a concern for simple or low-level projects, but for big Qt applications I worry about contamination problems.

First I will attempt to bootstrap CMake from a shell script, build CMake with CMake, and then look into what CMake helpers should be included with the SDK to make Qt work.

ehmry, 2019-03-20:
To clarify, Qbs is deprecated and Qt upstream will focus on QMake and CMake: https://blog.qt.io/blog/2018/10/29/deprecation-of-qbs/.

ehmry, 2019-03-21:
A native port is not viable, CMake depends on Libuv, which I assume it uses to spawn subprocesses. This is not going to work with the libc yet, and isn't worth the effort of making it work with Noux right now.

nfeske, 2019-03-21:
So you will follow the cross-compile route then?

ehmry, 2019-03-21:
Yes, the plan was always to cross-compile, but I will do that first.

ttcoder, 2019-03-21:
Disregard this comment if it's off-base, but I've checked out the haiku recipe for CMake out of curiosity....

https://github.com/haikuports/haikuports/blob/master/dev-util/cmake/cmake-3.9.5.recipe

and I see it disables (?) libuv:

-DCMAKE_USE_LIBUV=0

Though Haiku is a "commited to *nix and fork" type of OS... So their having a working recipe does not rule out the existence of other hidden noux-isms beside libuv.

ehmry, 2019-03-22:
I believe its a flag to use the system Libuv or the bundled uv.

ehmry, 2019-12-21:
Stale issue.

closed 2019-12-21 by ehmry