genodelabs/genode #3240: sculpt ce: lock caused by page fault in init -> leitzentrale -> manager

tomga, 2019-03-20:
During playing with sculpt ce I found a scenario where I can make system unusable.

1. + -> Depot -> cnuke -> Virtual Machines -> download debian
2. + (to hide)
3. disable "Use" of current filesystem
4. enable "Use" of filesystem
5. + - opens menu with download debian and clicking it causes page fault

After this the only thing that worked for me was to switch to wm once.

In case there is a problem with description I have a video with this I can put somewhere.

nfeske, 2019-03-21:
Thanks for reporting this issue. I immediately tried to reproduce it using the RAM fs, but I could trigger the fault. I'll continue trying. If you happen to come up with a way to trigger the issue in a reliable way starting with an empty ram fs, I would be very grateful.

tomga, 2019-03-21:
It was important to hide menu in state when opening it again would show menu deep in depot hierarchy and later disable and enable used filesystem. Opening again showed menu with package directly instead of browsing from root.

I can try to reproduce with RAM in the evening.

I emailed you video - mayby it will be enough. Yesterday I spent an hour to reproduce this reliably. Maybe you do something differently.

nfeske, 2019-03-21:
Thanks for the screen capture. I tried the steps several times (using disk and ram as fs), but no crash for me :-/

tomga, 2019-03-21:
I couldn't reproduce this with RAM filesystem either.

To let you diagnose this further I've prepared qemu disk image with content from my ThinkPad where I could reproduce problem. I've checked and recorder that it is reproducible in qemu too.

Qemu image with video can be downloaded (for next 7 days) from link below. After start press Esc for boot menu and start from number 3 (sculpt image).

https://send.firefox.com/download/be4d63e503/#QM9sYVpCYmKoj3FoV4uEiA

nfeske, 2019-03-22:
Thanks a lot @tomga for your engagement. I downloaded the archive and will continue my investigation after the weekend.

nfeske, 2019-03-26:
Just a quick status, thanks to your image, @tomga, I can reproduce the problem manually now. :-) Investigating...

nfeske, 2019-03-26:
I found and fixed the issue in https://github.com/genodelabs/genode/commit/f346bfa740f93ead3efc053d245900cc9cf48865. Thanks again @tomga for your splendid support! :-)

commit f346bfa7 (2019-03-26): sculpt manager: prevent double destruction

The '_currently_constructed' pointer caches the information about which
'Launched_child' is currently configured in the menu. When discarding
the runtime (e.g., when un-using a file system) at this point, this
cached pointer was not invalidated while all 'Launched_child' objects
would be freed (including the currently constructed one). On the next
attempt to construct a new child, the sculpt manager attempted to
destruct the 'Launched_child' referred by the (now outdated)
'_currently_constructed' again.

Fixes #3240
closed 2019-04-02 by nfeske