genodelabs/genode #3241: sculpt ce: installed package can stay in "installing" state in menu forever

tomga, 2019-03-20:
Attempt to install "download debian" package from cnuke's depot ends with:
```
[runtime -> update -> manager] Installation complete.
```
in the log but in menu it shows "installing..." forever and it isn't possible to start instance - it is not possible to assign "vm_fs".

Possibly it can be a result of earlier logs:
```
[runtime -> update] Warning: dynamic assigned RAM (24M) exceeds available RAM (24060830)
[runtime -> update] Error: RAM preservation exceeds available memory
```

I think that there is a problem with a package which triggers another with error handling.

nfeske, 2019-03-21:
I can reproduce this issue when starting with an empty ram fs. For some reason, the sculpt manager fails to interpret the state report of the update subsystem in the correct way. I'll investigate.

nfeske, 2019-03-21:
The problem is that @cnuke's pkg has a superfluous `<rom>` listed as its content in the runtime definition.

@cnuke could you please remove the `pthread.lib.so` entry and update the pkg and index? The pthread.lib.so does not exist anymore because the functionality is now part of the libc. The sculpt manager presents the configuration dialog only if the package is complete. But since the pthread.lib.so won't ever be installed, it remains incomplete even after the update mechanism has done its job.

tomga, 2019-03-21:
Shouldn't it be presented as "malformed package" or "installation error" or something like this?

nfeske, 2019-03-21:
Yeah. You are completely right.

valerius2k, 2019-03-21:
I acknowledge that this problem is caused by unneeded ROM modules. Yesterday, I updated my working copy from the repository, and encountered the same problem @tomga has. The same problem occurred with my own program. I had three unneeded ROM's. When I commented them out, my program now started successfully again.

cnuke, 2019-03-21:
Done.

nfeske, 2019-03-21:
Commit https://github.com/genodelabs/genode/commit/754a5959069a09647a7a212e225ea6b2f4dd3c33 implements @tomga's suggestion.

ehmry, 2019-03-22:
Is it possible to find out which ROMs are blocking a package? This issue is blocking the flif_capture package.

nfeske, 2019-03-22:
Yes, you can look in the blueprint of the queries package (/report/runtime/depot_query/blueprint) just after the failed installation. Look out for <missing_rom> nodes.

ehmry, 2019-03-22:
Excellent, thanks.

nfeske, 2019-04-02:
Fixed in master.

closed 2019-04-02 by nfeske