genodelabs/genode #3242: fs_tool: the test regularily fails on some platforms

skalk, 2019-03-21:
This night it failed on Fiasco.OC and Sel4 on x86_64 hardware and on Fiasco.OC Arndale board.

nfeske, 2019-03-21:
According to the output for Fiasco.OC, this is false negative.

I suspect that this issue is connected to #3238. When I created the test originally, I struggled with reliably matching the log output when including the (expected) warning messages in the pattern. So I left out the matching of the warnings in my original commit. Once I told @m-stein about it, he apparently managed to match the warnings and created the fixup https://github.com/nfeske/genode/commit/ad051253e8c8621a62063bceb681eb947e241fbf. But maybe he was just lucky when testing his version?

Anyway, to validate/dispel this claim, I added the reverted version of the @m-stein's fixup commit to staging (https://github.com/genodelabs/genode/commit/fe1b3a8272532ecb287b93fe60ffdaa65b5d344f) to see if the test stabilizes then. If so, the fs_tool may be a good test candidate for investigating #3238.

m-stein, 2019-03-21:
Interesting... I'm at it.

nfeske, 2019-03-21:
@m-stein thank you for making this a priority.

When looking again at the fixup for the fs_tool test, I think we should remove the wildcards at the end of the lines again. I understood your intention regarding the resilience against interwoven log output produced by unrelated components. But in this particular test, the underlying system is not expected to produce any intermediate lines of log output. Should that nevertheless happen, we should have a closer look.

m-stein, 2019-03-21:
This is fine with me.

skalk, 2019-03-22:
As far as I read the runtime recipe correctly, the test on Arndale is not a false-positive, on x86_64 it is indeed. On Arndale, the final `report_rom` message is missing.

nfeske, 2019-03-29:
Commit e04668eefb752c6daf9b267f67a239851a039198 makes the output deterministic.

nfeske, 2019-04-02:
Fixed in master.

closed 2019-04-02 by nfeske