genodelabs/genode #3243: driver_manager: increase block driver memory

alex-ab, 2019-03-21:


nfeske, 2019-03-21:
What is the background of this issue?

alex-ab, 2019-03-21:
The ahci driver requests memory on sculpt ce (T460) and the system hangs.

nfeske, 2019-03-21:
I wonder what's the difference to systems like mine (also using ahci) where this memory is not needed. Do you happen to have multiple AHCI devices connected?

Intuitively, assigning 10 MiB to a disk driver seem already quite generous. I have nothing against your patch but I'd like to understand why we need to bump up this value even higher (14 MiB according to your patch).

alex-ab, 2019-03-21:
> Do you happen to have multiple AHCI devices connected?

No.

I will have to take time to investigate since this is my working machine. I will report back when I knew more.

closed 2019-05-09 by alex-ab