genodelabs/genode #3245: depot_autopilot, nova, x86_64: recent failures

m-stein, 2019-03-21:
Since the last night, there are 36 test failing on this platform with qemu. It starts with test-part_block_mbr which produces log output but times out, then the consecutive tests all fail without any log output. But the system around the test runtime remains running just fine.

m-stein, 2019-03-21:
I try to bisect it back on a second machine.

m-stein, 2019-03-22:
This symptom is recurrent.
* It always occurs on nova+x86_64+qemu.
* It always starts at a specific test and causes all following tests to fail and refuse any log output.
* The first failing test might or might not be able to produce log output up to a certain point.
* The system outside the test runtime always keeps running just fine until the depot autopilot prints its result summary.

This is a list of when and how it occured since February:
```
date        nr of failing  first failing 
2019-03-21  36             test-part_block_mbr - 
2019-03-12  53             test-libc_connect_vfs_server_lx
2019-02-23  44             test-libc_vfs_counter
2019-02-19  48             test-libc_vfs
2019-02-18  53             test-libc_connect_vfs_server_lw
2019-02-01  62             test-fs_rom_update_fs
```

m-stein, 2019-03-22:
I'd suggest to add diagnostics to the system that are meant only for staging, just as we do with issue  #3238 but, so far, I have no plan what to add.

chelmuth, 2019-03-22:
Interestingly *test-part_block_mbr* failed to finish this night on *x86/x86_32*...

rite, 2019-06-18:
> Interestingly _test-part_block_mbr_ failed to finish this night on _x86/x86_32_...

Same here (x86_64/nova) while executing `run/depot_autopilot` `until_failed`. It succeeded 5 times and than failed. Please note that in contrast to @m-stein comment, in this particular case the following tests succeed nevertheless:

```
[14:19:16.835] [init -> depot_autopilot] --- Run "test-part_block_mbr" (max 60 sec) ---
[14:19:16.835] [init -> depot_autopilot] 
[14:19:17.199] [init -> depot_autopilot] 0.362 [init] parent provides
[14:19:17.200] [init -> depot_autopilot] 0.362 [init]   service "ROM"
[14:19:17.201] [init -> depot_autopilot] 0.363 [init]   service "IRQ"
[14:19:17.201] [init -> depot_autopilot] 0.364 [init]   service "IO_MEM"
[14:19:17.202] [init -> depot_autopilot] 0.365 [init]   service "IO_PORT"
[14:19:17.203] [init -> depot_autopilot] 0.366 [init]   service "PD"
[14:19:17.203] [init -> depot_autopilot] 0.366 [init]   service "RM"
[14:19:17.204] [init -> depot_autopilot] 0.367 [init]   service "CPU"
[14:19:17.205] [init -> depot_autopilot] 0.367 [init]   service "LOG"
[14:19:17.205] [init -> depot_autopilot] 0.368 [init]   service "Timer"
[14:19:17.226] [init -> depot_autopilot] 0.388 [init] child "rom_block"
[14:19:17.226] [init -> depot_autopilot] 0.389 [init]   RAM quota:  32520K
[14:19:17.227] [init -> depot_autopilot] 0.389 [init]   cap quota:  68
[14:19:17.227] [init -> depot_autopilot] 0.390 [init]   ELF binary: rom_block
[14:19:17.228] [init -> depot_autopilot] 0.390 [init]   priority:   0
[14:19:17.229] [init -> depot_autopilot] 0.392 [init]   provides service Block
[14:19:17.240] [init -> depot_autopilot] 0.403 [init] child "part_block"
[14:19:17.241] [init -> depot_autopilot] 0.403 [init]   RAM quota:  9992K
[14:19:17.241] [init -> depot_autopilot] 0.404 [init]   cap quota:  68
[14:19:17.242] [init -> depot_autopilot] 0.404 [init]   ELF binary: part_block
[14:19:17.242] [init -> depot_autopilot] 0.405 [init]   priority:   0
[14:19:17.243] [init -> depot_autopilot] 0.406 [init]   provides service Block
[14:19:17.251] [init -> depot_autopilot] 0.414 [init] child "report_rom"
[14:19:17.252] [init -> depot_autopilot] 0.414 [init]   RAM quota:  4872K
[14:19:17.252] [init -> depot_autopilot] 0.414 [init]   cap quota:  68
[14:19:17.252] [init -> depot_autopilot] 0.415 [init]   ELF binary: report_rom
[14:19:17.253] [init -> depot_autopilot] 0.415 [init]   priority:   0
[14:19:17.254] [init -> depot_autopilot] 0.416 [init]   provides service Report
[14:19:17.254] [init -> depot_autopilot] 0.417 [init]   provides service ROM
[14:19:17.266] [init -> depot_autopilot] 0.428 [init] child "test-part1"
[14:19:17.266] [init -> depot_autopilot] 0.428 [init]   RAM quota:  4872K
[14:19:17.267] [init -> depot_autopilot] 0.429 [init]   cap quota:  68
[14:19:17.267] [init -> depot_autopilot] 0.429 [init]   ELF binary: test-block-client
[14:19:17.268] [init -> depot_autopilot] 0.430 [init]   priority:   0
[14:19:17.276] [init -> depot_autopilot] 0.438 [init] child "test-part2"
[14:19:17.276] [init -> depot_autopilot] 0.439 [init]   RAM quota:  4872K
[14:19:17.277] [init -> depot_autopilot] 0.439 [init]   cap quota:  68
[14:19:17.277] [init -> depot_autopilot] 0.440 [init]   ELF binary: test-block-client
[14:19:17.278] [init -> depot_autopilot] 0.440 [init]   priority:   0
[14:19:17.286] [init -> depot_autopilot] 0.448 [init] child "test-part3"
[14:19:17.286] [init -> depot_autopilot] 0.449 [init]   RAM quota:  4872K
[14:19:17.286] [init -> depot_autopilot] 0.449 [init]   cap quota:  68
[14:19:17.287] [init -> depot_autopilot] 0.449 [init]   ELF binary: test-block-client
[14:19:17.287] [init -> depot_autopilot] 0.450 [init]   priority:   0
[14:19:18.203] [init -> depot_autopilot] 1.362 [init -> test-part2] Warning: Don't call Genode::Env::exec_static_constructors() in components without static s
[14:19:18.216] [init -> depot_autopilot] 1.376 [init] child "report_rom" announces service "Report"
[14:19:18.217] [init -> depot_autopilot] 1.377 [init] child "report_rom" announces service "ROM"
[14:19:18.226] [init -> depot_autopilot] 1.386 [init] child "rom_block" announces service "Block"
[14:19:18.230] [init -> depot_autopilot] 1.388 [init -> test-part1] Warning: Don't call Genode::Env::exec_static_constructors() in components without static s
[14:19:18.257] [init -> depot_autopilot] 1.416 [init -> rom_block] using file=ata.mbr.raw as device with block size 512.
[14:19:18.260] [init -> depot_autopilot] 1.419 [init -> test-part3] Warning: Don't call Genode::Env::exec_static_constructors() in components without static s
[14:19:18.595] [init -> depot_autopilot] 1.752 [init -> part_block] Partition 1: LBA 2048 (512 blocks) type: 83
[14:19:18.596] [init -> depot_autopilot] 1.753 [init -> part_block] Partition 4: LBA 4096 (16384 blocks) type: 5
[14:19:18.597] [init -> depot_autopilot] 1.755 [init -> part_block] Partition 5: LBA 6144 (256 blocks) type: 7
[14:19:18.599] [init -> depot_autopilot] 1.756 [init -> part_block] Partition 6: LBA 10240 (64 blocks) type: 83
[14:19:18.600] [init -> depot_autopilot] 1.757 [init -> part_block] Partition 7: LBA 14336 (512 blocks) type: 83
[14:19:18.601] [init -> depot_autopilot] 1.759 [init -> part_block] Partition 8: LBA 18432 (128 blocks) type: 83
[14:19:18.602] [init -> depot_autopilot] 1.760 [init -> part_block] Partition 9: LBA 8448 (256 blocks) type: 83
[14:19:18.610] [init -> depot_autopilot] 1.767 [init -> report_rom] report 'part_block -> partitions'
[... cut ...]
[14:23:17.075] [init -> depot_autopilot]  test-weak_ptr                   ok         3.923  log "--- finished test-weak_ptr ---"
[14:23:17.076] [init -> depot_autopilot]  test-xml_node                   ok         0.443  log "[init -> test-xml_node] --- XML- ..."
[14:23:17.076] [init -> depot_autopilot]  gcov                            ok        11.204  log "[init] child "gcov" exited with  ..."
[14:23:17.077] [init -> depot_autopilot] 
[14:23:17.077] [init -> depot_autopilot] succeeded: 73 failed: 1 skipped: 4
```

rite, 2019-06-18:
Argh, I meant to run it with the debug commits mentioned in #3238 in place but took the wrong branch... I'll try to reproduce it again...

rite, 2019-06-19:
Ok, now with debug output - complete log at [1]. `test-part_block_mbr` failed in the 8th iteration (start at line 42138). Interestingly, `test-resource_yield` failed in the same iteration (start at line 43075).

---
[1] https://gist.github.com/rite/f4c745fbc26e6a87d2ce7bd73e7ef2b3

chelmuth, 2019-09-26:
I accidentally stumbled over this issues on a search for *constructor* and was wondering which failing tests the issue may include. Is it still valid? Could we change the issue title to be more specific about the actual failing test(s)?

rite, 2019-09-26:
@chelmuth I didn't encounter any problem with depot_autopilot on x86_64/nova since, I think, around https://github.com/genodelabs/genode/releases/tag/sculpt-19.07

closed 2019-09-27 by chelmuth