genodelabs/genode #3248: fetchurl_lxip: hangs sporadically

skalk, 2019-03-22:
Every night, we can see the `fetchurl_lxip` test to hang on several platforms. Moreover, some of us already experienced a hanging fetchurl when installing packages in Sculpt. Maybe, that hits the same issue.

ehmry, 2019-03-22:
For the record I always prefer lwip over lxip.

nfeske, 2019-03-22:
Why must this information be recorded? ;-)

ehmry, 2019-03-22:
Lwip has a worse reputation than it deserves.

cnuke, 2019-05-08:
A similar behaviour is observable in the *genodians.org* appliance where sometimes a download is stuck (no progress is made which can be checked by looking at the progress report) until eventually some internal timeout is triggered and the download attempt is aborted.

nfeske, 2021-02-26:
This issue has become stale. I'm closing it.

closed 2021-02-26 by nfeske