genodelabs/genode #3249: rpi: fs_packet test does not succeed in time

skalk, 2019-03-22:


skalk, 2019-03-22:
Although, we increased the timeout of this test already to 230 seconds it does not succeed in time on Raspberry Pi 1. It is not starving, so it still makes progress, but it seems to be quite slow.
It is not solely related to the kernel, but clearly base-hw does not shine regarding this test:

kernel | x86_64 | imx6q | rpi
---------|----------|----------|-----
hw      | 22s       | 107s   | >250s
foc      | 9s         | 113s   | -
sel4    | 6s         | 85s     | -
I'm not sure, what to make out of this. Shall we investigate why that test is that slow on those platform, or just again increase its timeout? Any opinions?

m-stein, 2019-03-25:
I wonder whether RPI1 is worth investigation regarding a potentially platform-specific performance issue while we're already talking about RPI3 on the mailing list?

nfeske, 2019-03-25:
@m-stein According to my experience, weaker platforms are especially useful for investigating performance issues because they are more unforgiving. The observations of @skalk have most certainly not much to do with "our rpi support" but they instead point at a systemic issue.

m-stein, 2019-03-25:
@nfeske Looking at the imxq6 result of hw, I didn't get the feeling that the issue most certainly hasn't much to do with "our rpi support".

tomga, 2019-03-25:
@m-stein Currently sold Rpi Zero models use bcm2835 chip - the same as in Rpi1 so work wouldn't be wasted. If/when I succeed with rpi3 I'd like to work on support for other models.

@skalk Is there a reason why you did not put result for rpi on foc? I've seen that support for rpi on foc was available. Maybe this would help to diagnose the problem?

skalk, 2019-03-26:
@tomga the reason is that we currently do not test Fiasco.OC/rpi in our nightly infrastructure. Support for rpi on foc was a community effort several years ago, but nobody was actively using it for a longer time. And nobody actively maintained it. Actually, I wanted to move that platform (I mean kernel/board combination) and some more - all that are not regularily tested in our Lab - into genode-world. Thereby, it would be transparent to everybody what platform is tested in Genode Labs, and which not. Anyway, if it now gets alive again, we should think about re-activating Fiasco.OC/rpi in our nightly tests, and move to rpi3 as soon as we have it mainline.

tomga, 2019-03-26:
I didn't know that support for Fiasco.OC/rpi was a community effort and that it was not maintained and not tested regurlarly. I've seen on mailing list last year that it was working at that moment and thought that results could help with diagnosing this problem.

skalk, 2019-03-26:
Well, I expect it to work in general, because AFAIK there are no rpi specific corners in our base-foc support. So if you like to measure differences to hw kernel I think it is possible. The actual support for rpi 1 was brought in by Reinier Millo Sánchez in 2015. I thought most discussion in the past month on the mailing list were efforts to run on rpi 3, but maybe I've missed something.

tomga, 2019-03-26:
Yes, currently I'm trying to make changes for rpi 3 but I'd like to work on rpi support for other models later as they don't seem to differ much.

tomga, 2019-04-01:
Not sure if it helps but on rpi3bplus this test finishes in 133 seconds. Given that processor frequency is two times slower in rpi1 then rpi1 doesn't need much more to finish. On the other hand processor in rpi3bplus has higher frequency but worse test performance than IMX6. Probably it will need to be investigated in the future.

skalk, 2019-04-02:
@tomga thank you for your insights regarding rpi3. I'm not sure how they relate to the rest. Currently, I enable rpi3 for Genode too. As first low-hanging-fruit I address Fiasco.OC. Let's see what we get there, and whether it is related to the hw-kernel, like numbers suggests that I've posted before.

skalk, 2020-03-03:
Ok, due to the recent performance improvements on the hw kernel, the test only needs up to two seconds maximum across all platforms including rpi 1. I therefore re-enabled the test for rpi, and mark this as fixed.

nfeske, 2020-03-03:
That is so cool!

tomga, 2020-03-03:
> up to two seconds maximum across all platforms including rpi 1.

@skalk did you mean minutes?

skalk, 2020-03-03:
@tomga nope, I meant seconds

tomga, 2020-03-03:
@skalk, now I'm very puzzled. I expected quick answer: yes and I would assume that optimizations in hw made it more than two times faster on rpi1 and that it would be a really, really great achievement. 

Now, given times you measured year ago, you say that it is now about 125 times faster on rpi1. Does this test really measure the same thing? And what about times on other kernels that probably weren't optimized?

skalk, 2020-03-03:
Ahh, I've found the issue #3562, which was responsible for the poor performance and already fixed the problem. Anyway, now hw is faster regarding the test than sel4 and foc ;-).

tomga, 2020-03-03:
Thanks for explanation. :-)

chelmuth, 2020-03-11:
Merged to staging.

commit f954ab63 (2020-03-11): depot_autopilot: re-enable test-fs_packet on rpi

Fixes #3249
closed 2020-03-26 by nfeske