genodelabs/genode #3251: add support for i.MX7 Dual SABRE reference board

skalk, 2019-03-22:


nfeske, 2019-03-25:
Thanks for the beautiful commit series, which I just merged to staging.

nfeske, 2019-03-26:
There is a build problem with the depot archive bin/x86_64/base-hw-muen:
```
/data/genode/repos/base/../base-hw/src/core/kernel/vm_thread_on.cc: In member function âvoid Kernel::Thread::_call_new_vm()â:
/data/genode/repos/base/../base-hw/src/core/kernel/vm_thread_on.cc:27:17: error: âCpu_state_modesâ is not a member of âGenodeâ
  _call_new<Vm>((Genode::Cpu_state_modes*)user_arg_2(), context,
                 ^~~~~~
/data/genode/repos/base/../base-hw/src/core/kernel/vm_thread_on.cc:27:41: error: expected primary-expression before â)â token
  _call_new<Vm>((Genode::Cpu_state_modes*)user_arg_2(), context,
                                         ^
    COMPILE  region_map_component.o
    COMPILE  region_map_support.o
/data/genode/repos/base/mk/generic.mk:56: recipe for target 'kernel/vm_thread_on.o' failed
make[6]: *** [kernel/vm_thread_on.o] Error 1
```

From a distance, it seems to be caused by https://github.com/genodelabs/genode/commit/45eb50573305d443670ce7c61f6e243eb28a27d2.

skalk, 2019-03-26:
@nfeske thanks for reporting, I'll have a look

skalk, 2019-03-26:
@nfeske here is a fixup commit 8526348e797c75c12bfa6c390b49c2e4b212ab76 for Muen

nfeske, 2019-03-26:
Thanks Stefan!

nfeske, 2019-03-27:
Unfortunately, I missed to report another build problem yesterday. @skalk could you please have a look?

```
/data/genode/repos/base-sel4/recipes/src/base-sel4-imx6q_sabrelite/content.mk:30: *** Error: sel4_elfloader port does not exist.  Stop.
/data/genode/tool/depot/mk/extract_content.inc:51: recipe for target '/data/depot/genodelabs/src/base-sel4-imx6q_sabrelite/incomplete/LICENSE' failed
make[3]: *** [/data/depot/genodelabs/src/base-sel4-imx6q_sabrelite/incomplete/LICENSE] Error 1
bash: line 12: /data/depot/genodelabs/src/base-sel4-imx6q_sabrelite/incomplete/used_apis: No such file or directory
make[3]: Target 'base-sel4-imx6q_sabrelite' not remade because of errors.
/data/depot/var/extract.mk:24: recipe for target 'genodelabs/src/base-sel4-imx6q_sabrelite/2019-03-17' failed
make[2]: *** [genodelabs/src/base-sel4-imx6q_sabrelite/2019-03-17] Error 2
make[2]: Target 'all' not remade because of errors.
/data/genode/tool/depot/extract:180: recipe for target 'execute_generated_extract_mk_file' failed
make[1]: *** [execute_generated_extract_mk_file] Error 2
make[1]: Target 'genodelabs/src/base-sel4-imx6q_sabrelite' not remade because of errors.
/data/genode/tool/depot/create:41: recipe for target 'extract' failed
make: *** [extract] Error 2
make: Target 'genodelabs/src/base-sel4-imx6q_sabrelite' not remade because of errors.
```

skalk, 2019-03-27:
@nfeske I still often forget the packages, here is a fixup commit for it f6a6b0b7871a94aace45ca8f30f8c1b0d25b144d

nfeske, 2019-03-27:
No problem @skalk. Thanks for the fix. I just added it to staging.

commit 858764f9 (2019-03-25): hw: add support for i.MX7 Dual SABRE board

Fix #3251
closed 2019-04-02 by nfeske