genodelabs/genode #3253: nova: '.init_array' not sorted correctly

cproc, 2019-03-25:
When trying to link the NOVA kernel with binutils 2.32, the linker reports syntax errors in the linker script (hypervisor.ld) for these lines:

```
KEEP (SORT_BY_INIT_PRIORITY(*)(.init_array.65534 .ctors.00001))
KEEP (SORT_BY_INIT_PRIORITY(*)(.init_array.65533 .ctors.00002))
KEEP (SORT_BY_INIT_PRIORITY(*)(.init_array.* .ctors.*))
SORT_BY_ALIGNMENT (*)(.cpulocal.hot)
SORT_BY_ALIGNMENT (*)(.cpulocal)
```

There has been a binutils commit [1] which mentions that `SORT_BY_ALIGNMENT` and `SORT_BY_INIT_PRIORITY` for file names had previously no effect and are now considered an error.

When building the kernel with the current tool chain and adding the `-Wl,-M` flag to the link command line to print a link map, it indeed shows that the constructors are not sorted according to their priorities (65434, 65433):

```
...
*(.init_array.* .ctors.*)
.ctors.65434   0xffffffff81024df0        0x8 buddy.o
.ctors.65433   0xffffffff81024df8        0x8 dmar.o
.ctors.65433   0xffffffff81024e00        0x8 hpet.o
.ctors.65433   0xffffffff81024e08        0x8 ioapic.o
.ctors.65433   0xffffffff81024e10        0x8 mtrr.o
.ctors.65433   0xffffffff81024e18        0x8 pci.o
.ctors.65434   0xffffffff81024e20        0x8 pd.o
.ctors.65433   0xffffffff81024e28        0x8 pd.o
...
```

After modifying the linker script in commit [2] the sorting seems to work:

```
...
*(SORT(.init_array.*) SORT(.ctors.*))
.ctors.65434   0xffffffff81024df0        0x8 buddy.o
.ctors.65434   0xffffffff81024df8        0x8 pd.o
.ctors.65433   0xffffffff81024e00        0x8 dmar.o
.ctors.65433   0xffffffff81024e08        0x8 hpet.o
.ctors.65433   0xffffffff81024e10        0x8 ioapic.o
.ctors.65433   0xffffffff81024e18        0x8 mtrr.o
.ctors.65433   0xffffffff81024e20        0x8 pci.o
.ctors.65433   0xffffffff81024e28        0x8 pd.o
...
```

but now the kernel does not boot anymore.

I assume that one or more of the priorities need to be adjusted in the NOVA source code, but I'm not sure which. Maybe @alex-ab can help here?

[1] https://github.com/bminor/binutils-gdb/commit/2b94abd48aef2d91bae1c35c8c10ebfb8757247d#diff-bbe014dacf19b0da545096357e00668e
[2] https://github.com/cproc/NOVA/commit/a6d90fa857794905a784c1e00c5aee1c4faf7144

alex-ab, 2019-04-02:
@cproc: thanks for the analyses - please try https://github.com/alex-ab/NOVA/commit/ae81da16023382bec80642376d77bf6927188125 with your new toolchain

cproc, 2019-04-04:
@alex-ab: it works with your commit

alex-ab, 2019-04-09:
@chelmuth: please consider for staging

chelmuth, 2019-05-06:
Fixed in master by 26796c8

closed 2019-05-06 by chelmuth