genodelabs/genode #3254: imx6q_sabrelite: No run script for depot_autopilot

m-stein, 2019-03-26:
The nightly depot-autopilot tests for this platform are not running and even not marked as missing because the run script is not found:
```
[2019-03-26 07:21:51] make[1]: Entering directory '.../imx6q_sabrelite/imx6q_sabrelite'
[2019-03-26 07:21:51] Error: No run script for depot_autopilot
[2019-03-26 07:21:51] Makefile:319: recipe for target 'run/depot_autopilot' failed
[2019-03-26 07:21:51] make[1]: *** [run/depot_autopilot] Error 255
[2019-03-26 07:21:51] make[1]: Leaving directory '.../imx6q_sabrelite/imx6q_sabrelite'
```

cnuke, 2019-03-26:
@nfeske please merge c7cb02f to staging.

nfeske, 2019-03-26:
Thanks. Merged to staging.

skalk, 2019-03-26:
@cnuke thank you for fixing the regression that I produced!

nfeske, 2019-04-02:
Fixed in master.

closed 2019-04-02 by nfeske