genodelabs/genode #3255: tool_chain: make sudo command configurable

cproc, 2019-03-26:
By default, the 'tool_chain' script installs the tool chain in '/usr/local/genode-gcc' using the 'sudo' command. To install the tool chain somewhere else (for example in a user home directory), it is already possible to override 'INSTALL_LOCATION' on the command line, but still 'sudo' is called, which might not be allowed on a multi-user system. A 'SUDO' variable which is 'sudo' by default and can be redefined as empty on the command line would solve this problem.

cproc, 2019-03-28:
I replaced '=' with '?=' at the suggested places. It didn't matter when putting the redefinitions after the command on the command line, but with '?=' it's also possible to put the redefinitions before the command or to have them in the environment.

nfeske, 2019-03-28:
Good. I merged your new commit to staging.

commit a9d99c8c (2019-03-28): tool_chain: make sudo command configurable

Fixes #3255
closed 2019-04-02 by nfeske