genodelabs/genode #3256: test.run: support init test

nfeske, 2019-03-27:
The current version of the test.run tool is not able to execute the init test because the test hosts its configuration in a dedicated ROM module provided by a raw archive, outside the pkg's runtime file.

To accommodate tests like this, the test.run tool must include the content of raw archives in the boot image. This is not possible in the general case because the content is generated according to the raw-archive's content.mk file. But for most tests - like test-init - the content.mk copies the content from `recipes/raw/<archive>/` (https://github.com/genodelabs/genode/blob/master/repos/os/recipes/raw/test-init/content.mk). So adding this convention to the heuristics of the test.run tool will bring us quite far.

skalk, 2019-03-28:
Cool, helpful thing, I experienced similar problems with another test package in the past, but did not had the time to investigate it further.

nfeske, 2019-04-02:
Fixed in master.

closed 2019-04-02 by nfeske