genodelabs/genode #3257: Replace global VFS response handlers with local handlers

ehmry, 2019-03-27:
The VFS and libc current operate together using a global I/O response handler. This is fine for the libc, but complicates the VFS server and some of the VFS plugins because a reference to the global handler must be passed around to the to notification sites.

Allowing the application to create multiple handlers and attach them to handles simplifiles the plugins and the server. Sending ready-ready and IO-progress notifications through handle-specific channels also makes it easier to monitor the behavior of the VFS.

ehmry, 2019-03-27:
I have a patch at 5067859d2713011a4e821093c60e571cb6c0eb2e that replaces the global I/O and watch handles and removes the last use of the `Post_signal_hook`. This is a big patch because it was necessary to refactor how the VFS server processes pending packets, but as a result the VFS server can process multiple packets on a single session by strongly associating pending packets with open handles. This patch is a net reduction in code and I hope it makes the VFS easier to understand, but it shouldn't significantly improve performance. However, a simpler VFS is easier to instrument and profile, which I see as the next step now that `Post_signal_hook` is gone.

ehmry, 2019-03-27:
Noux is unconverted, but I just need to rebase an old patch and test.

Ref #3252

nfeske, 2019-03-27:
I browsed though the patch and it indeed is a substantial improvement. It untangles the parts that I liked the least about the VFS. Thanks @ehmry for this massive effort!

m-stein, 2019-03-27:
:+1: It's cool to see the VFS getting substantially simplified!

ssumpf, 2019-03-27:
Wow!

ehmry, 2019-04-01:
I have a documentation fixup at cd9a78559f03795f81257d6d1ec8e994d4bdb45c, the Noux conversion at 7022dd41db51a0fe86a0edce17fa9a8a41d88fc5, and an lxip fix at 4d166041109c18894db7712bfd2251b5704504e1.

ehmry, 2019-04-01:
Lwip fixup at cc519df21aac70c7993d10aa0d5e8474e6315007

ehmry, 2019-04-02:
Fixup for menu_view at 39297357e5e2e8a988f2cc2c9bd4d44b75a607a9.

nfeske, 2019-04-02:
I merged the commits squashed into one to staging now.

nfeske, 2019-04-02:
The commenting of the VFS is great. It is quite a pleasure to read the code now.

ehmry, 2019-04-02:
I have a fixup for the watch handler accessor at c4c8f145b814e9a9283281d635e76005d037372c, its not used anywhere so its easy to fix.

ehmry, 2019-04-02:
Or it can be removed with 03db95f6f53b189cf4d62f5da1c40b75a2398d6d.

nfeske, 2019-04-02:
Thanks Emery for the fixups, which I squashed to ca113b740b14e55f74e9faf86e67fa7920cf649a.

nfeske, 2019-04-03:
According to the nightly tests, the following tests are worth looking at:

* rump_ext2 on many platforms, e.g., NOVA on Qemu
* test-libc_vfs on OKL4 on Qemu (depot autopilot)

nfeske, 2019-04-03:
* depot_download on NOVA on Qemu is affected similarly, with lots of messages like this:
  ```
  [2019-04-03 06:37:55] [init -> depot_download -> dynamic -> extract] [34mWarning: ack for unknown File_system handle 206[0m[0m
  ```

nfeske, 2019-04-03:
Not sure if related, but in the cpu_sampler_noux test on NOVA on Qemu/x86_64, Noux dies with the following message:
```
[2019-04-03 05:24:37] attempted write at read-only memory (WRITE pf_addr=0xaa020 pf_ip=0x83481 from pager_object: pd='init -> noux' thread='ep') [0m

[2019-04-03 05:24:37] page fault, pd='init -> noux' thread='ep' cpu=0 ip=0x83481 address=0xaa020 stack pointer=0xa02fef18 qualifiers=0x7 irUWP reason=2[0m
```

nfeske, 2019-04-03:
The noux_tool_chain_auto.run script is now broken on all platforms. On NOVA/x86_64, the following message appears in the log:

```
[2019-04-03 05:15:33] attempted write at read-only memory (WRITE pf_addr=0x10454c7 pf_ip=0x83481 from pager_object: pd='init -> noux' thread='ep') [0m

[2019-04-03 05:15:33] page fault, pd='init -> noux' thread='ep' cpu=0 ip=0x83481 address=0x10454c7 stack pointer=0xa02fef18 qualifiers=0x7 irUWP reason=2[0m
```

Looks similar (same IP) to the one of cpu_sampler_noux.

nfeske, 2019-04-03:
In the vfs_import.run test on NOVA/x86_64, noux apparently gets stuck.

nfeske, 2019-04-03:
As a good news, there is not a single build error. ;-)

ehmry, 2019-04-03:
I suppose the other good news is that the tests work...

Fixup for Noux at ec8e562c47eabcbe3a4862fb93a067889d44e077.
Fixup for unknown handles at 0934e43b1d3f19988fc60f0aed9df5de67efecda.

nfeske, 2019-04-03:
Good news indeed. Merged to staging.

Do you have a plan for the XXX that you added to the libc, @ehmry?

ehmry, 2019-04-03:
I would consider adding a flag that marks a handle as written, otherwise I would not want a sync packet round trip when opening a file to test for its presence.

Fixup for lwIP at 911ca56267b531c4c7e9709b87e6c70ff275fc96.

nfeske, 2019-04-03:
Thanks a lot, @ehmry!

> I would consider adding a flag that marks a handle as written, otherwise I would not want a sync packet round trip when opening a file to test for its presence.

That's also my line of thinking. Right now, we'd get the round-trip for each close, right?

ehmry, 2019-04-03:
Yes, and `int access(const char *path, int mode)` is deprecated, so applications should be using open.

ehmry, 2019-04-04:
I'm worried that the changes to Noux are too drastic would like to partially revert them, fixup at 341ee291e18be2906b1f3d900140f61ffee8e82d.

commit 29df9453 (2019-04-02): VFS: Replace global response handlers with local handlers

Replace the I/O response handler that is passed to the VFS at
construction with an object that is dynamically attached to handles.
This object shall also accept read-ready notifications, and plugins are
encouraged to keep handles awaiting ready-ready notifications separate
from handles that await I/O progress.

Replace the use of handle lists in plugins with handle queues, this
makes the code easier to understand and the ordering of notifications to
the application more explicit.

These changes replace the use of the Post_signal_hook f [... cut]
commit 85816184 (2019-04-03): VFS: Replace global response handlers with local handlers

Replace the I/O response handler that is passed to the VFS at
construction with an object that is dynamically attached to handles.
This object shall also accept read-ready notifications, and plugins are
encouraged to keep handles awaiting ready-ready notifications separate
from handles that await I/O progress.

Replace the use of handle lists in plugins with handle queues, this
makes the code easier to understand and the ordering of notifications to
the application more explicit.

These changes replace the use of the Post_signal_hook f [... cut]
commit a6358735 (2019-04-09): VFS: Replace global response handlers with local handlers

Replace the I/O response handler that is passed to the VFS at
construction with an object that is dynamically attached to handles.
This object shall also accept read-ready notifications, and plugins are
encouraged to keep handles awaiting ready-ready notifications separate
from handles that await I/O progress.

Replace the use of handle lists in plugins with handle queues, this
makes the code easier to understand and the ordering of notifications to
the application more explicit.

These changes replace the use of the Post_signal_hook f [... cut]
closed 2019-04-09 by chelmuth