genodelabs/genode #3260: ARM 64-bit support

skalk, 2019-03-29:
It is time that Genode supports more recent ARM platforms again. To start with the development, I would first enable running Genode's userland on a kernel with existing support for Raspberry Pi 3 to run on top of Qemu. Next to run it on hardware. And lastly to support it in our hw kernel.

skalk, 2019-03-29:
For now, I can successfully boot a Genode/Fiasco.OC image on Qemu and Raspberry Pi 3 hardware. The the kernel complains about an alignment error in core within cmpxchg. I will investigate further.

ssumpf, 2019-04-03:
@skalk: Please let me know when the foc version reaches a state where I can enable dynamic linking support.

skalk, 2019-04-03:
@ssumpf I certainly will do so in the very near future ;-)

skalk, 2019-04-03:
@ssumpf ok, I'm done with bootstrapping core. It now likes to use ld.lib.so which obviously fails. You find my current work on my topic branch named issue#3260

ssumpf, 2019-04-03:
@skalk: Thanks, will start with dynamic linking immediately.

ssumpf, 2019-04-04:
I am happy to write that dynamic linking is working on ARM 64-bit. 09f767b and 78b6786dc8813a05fa4af469fde0383006799c13 will do the trick. In order to test this I have revived `ldso.run` with 6abefdaadeb0dfdd462e57f25e274175bc1efd9d (@nfeske, @chelmuth : would it be possible to get `ldso.run` back to staging, since it is such a low level component? If not, I can always cherry pick, so no problem). For the test to work a0e78389f2f3e01bae5e6991eecb893e3f9bbd77 needs also to be applied because there is no libc for armv8, yet. Also @skalk and myself fixed the DEBUG_CAP handling on base-foc with b85027e0ea13fdbad99f260bdd9894e7a45381ec, so please consider this one for staging.

nfeske, 2019-04-04:
Great news, @ssumpf!

> would it be possible to get ldso.run back to staging, since it is such a low level component?

Does the following command work for you?
```
make run/test PKG=test-ldso KERNEL=...
```

It uses the information from the test recipe but sidesteps the use of the depot, as described here: http://genodians.org/nfeske/2019-02-05-shortcut-for-testing

nfeske, 2019-04-04:
BTW, I just added the commit https://github.com/genodelabs/genode/commit/8b92dc217a537485809519f988d5b6d0a2b6e967 to staging, which makes test-ldso work completely with test.run, including the matching of the log output.

m-stein, 2019-04-04:
@ssumpf @skalk Wow, what an advance!

ssumpf, 2019-04-05:
@nfeske: 8b92dc2 works like a charm. Cool and thanks for the reminder.

skalk, 2019-04-05:
@ssumpf well, that was fast! Thanks for immediately stepping into the linker enablement.

skalk, 2019-05-20:
@ssumpf I've aligned our previous work on this issue with the current staging branch. You can find the result at my topic branch `issue#3260`

skalk, 2019-05-20:
@ssumpf: now there is a compilation issue with ld:
```
repos/base/src/lib/ldso/include/dynamic.h:339:5: error: ‘Plt_got’ was not declared in this scope
     Plt_got r(*_dep, _pltgot);
```

ssumpf, 2019-05-20:
@skalk: Thx, I know, will fix.

ssumpf, 2019-05-20:
@chelmuth: Please add ab99f57 to staging after the tool chain update, it makes ARM 64 compile.

skalk, 2019-05-29:
This issue is solved with current master state. The missing ARM64 support in the hw kernel should be topic of a dedicated issue.

closed 2019-05-29 by skalk