genodelabs/genode #3261: Convert  graphical terminal to Nitpicker client

ehmry, 2019-03-29:
I would volunteer. Ref #3196

ehmry, 2019-03-29:
I have an initial conversion patch 0d1761d54bfd00dd3f6ec2fd3557021ccf928a5c. There is more work to be done, but not without some deeper refactoring.

I've tested this on my Sculpt host, but I'm no longer able to use the runs scripts in gems, something is always out of sync between the recursive build systems and it takes too much time to work with.

ttcoder, 2019-03-31:
Off-topic question for @ehmry et al:

> , but I'm no longer able to use the runs scripts in gems, something is always out of sync between the recursive build systems and it takes too much time to work with.

Is the "Genode build system" a topic that is worth discussing, maybe off-ticket (on my @netcourrier.com addy ..etc) ? I have a reduced test case which preserves most of the great features of the Genode build system, yet starts compiling after a couple seconds of wait, by opposed to the 10-20 seconds turn-around time I see when using on gmake (invoking "make again" e.g.).

But it would take an awful lot of work to finish (at least for me), and I know you're working on a system of your own, different from what I have in mind; if so then disregard this

nfeske, 2019-04-01:
That's great @ehmry!

It would be nice to keep a way to defined the initial window size. We used to do that via the `initial_width` and `initial_height` attributes of `nit_fb`. How about introducing the attributes `initial_columns` and `initial_rows` to the terminal now? (not using pixel values keeps the configuration independent from the screen's DPI)

ehmry, 2019-04-01:
@nfeske, yes that would be a logical next step now that we have a baseline nitpicker client.

ehmry, 2019-04-02:
I have an additional commit that trims the terminal window down to a clean single-pixel border at c6c05b0cdc804c114e3de80215cf1a2c05dccf6c. I tried to include support for `initial_[columns|rows]` configuration, but it seems to be overriden by the window manager immediately.

nfeske, 2019-04-02:
@ehmry I have trouble reviewing your commit. It touches so many things. Some things seem to be on the account for different tastes while others are important. But I'm unable to keep those things sorted. 

According to my experience, the resize handling is a very delicate issue. I spent days to get it right. So I don't feel well with the complexity of the commit that touches this part of the code.

I would very much appreciate digestible commits with small incremental changes.

chelmuth, 2019-07-09:
Related to #3196.

nfeske, 2019-07-09:
The first commit https://github.com/ehmry/genode/commit/858f0062cd0ea183775a3d9c004685e38bfafdc1 looks good to me. But it cannot be merged in isolation as we would lose the setting of the initial window size.

The second one, I have a hard time swallowing - given the amount of testing and refinement that I invested into the current implementation I know that it is all too easy to introduce subtle problems (off-by-ones) especially related to resize operation. So when reading these invasive changes, I'm skeptical and would need to schedule some time for a thorough review.

To bring this issue gradually forward without me sitting on the critical path, I suggest adding the `<initial rows="80" columns="25/>` feature as a separate commit that can be used together with the first one.

ehmry, 2019-07-12:
I will yield to someone who can do this less invasively.

closed 2019-07-12 by ehmry