genodelabs/genode #3268: Shift Sculpt Leitzentralle CPU affinity

ehmry, 2019-04-01:
Would pinning the Leitzentralle and depot management components to a different CPU reduce the amount of scheduling interference between user-depoyed system and the administrative system? I assume that the first CPU has better performance than the second due to cross CPU RPC, but if the Leitzentralle is only used occasionally then I expect it would be fine to push it back to a second core?

nfeske, 2019-04-01:
Indeed. I vaguely remember that I already did this in the past (to a quite positive effect - at least regarding the fading). But I apparently missed to merge this change to upstream.

NobodyIII, 2019-05-02:
I'm interested in this as well. Has it been merged yet?

ehmry, 2019-05-02:
It probably won't happen until we start building Sculpt for the next release, which will happen towards the end of the month.

NobodyIII, 2019-05-02:
Ah. I'd like to see it in the 19.05 release if possible, but I can
understand the decision to wait.

nfeske, 2019-06-17:
Fixed in 1f96b4bc9a3136215a4a553d5c2a57e10351cd91.

commit 84a4d978 (2019-06-17): sculpt: use secondary CPU for leitzentrale

By decoupling the leitzentrale from the (contended) boot CPU, the fading
on F12 interferes much less with animations like nano3d deployed in the
runtime.

Fixes #3268
closed 2019-07-09 by chelmuth