genodelabs/genode #3269: lwIP nameserver file not available over File_system sessions

ehmry, 2019-04-01:
When the lwIP plugin is hosted by the VFS server the nameserver file cannot be opened by clients because the VFS server cannot open a handle on the root of the lwIP plugin.

ehmry, 2019-04-01:
Fixed at 9a9097ef707a588de4a5f245248445d04d4a8672.

ehmry, 2019-04-02:
I have a fixup for the unused arguments at b18692db99dc57789b7be1139bfb9880fde7577c.

nfeske, 2019-04-02:
Thanks @ehmry. I merged both commits as one to staging.

nfeske, 2019-04-03:
The nightly tests show trouble with netperf_lwip on many platforms (e.g., NOVA on hardware):

```
[2019-04-03 01:09:45] [init -> netserver_genode] configuring lwIP interface with DHCP[0m[0m

[2019-04-03 01:09:45] [init -> netserver_genode] [34mWarning: Don't call Genode::Env::exec_static_constructors() in components without static globals[0m[0m

[2019-04-03 01:09:52] [init -> netserver_genode] lwIP Nic interface up address=10.0.0.9 netmask=255.255.255.0 gateway=10.0.0.1[0m[0m

[2019-04-03 01:09:52] 
[2019-04-03 01:09:55] 
[2019-04-03 01:09:55] ---------------------------- TCP_STREAM -----------------------
[2019-04-03 01:09:55] spawn netperf-2.6.0 -4 -H 10.0.0.9 -P 1 -v 2 -t TCP_STREAM -c -C -- 1024 
[2019-04-03 01:09:55] establish control: are you sure there is a netserver listening on 10.0.0.9 at port 12865?
[2019-04-03 01:09:55] establish_control could not establish the control connection from 0.0.0.0 port 0 address family AF_INET to 10.0.0.9 port 12865 address family AF_INET
[2019-04-03 01:09:55] Error: Spawned process died unexpectedly
```

nfeske, 2019-04-03:
The nic_router.run script fails now on many platforms, e.g., NOVA on Qemu/x86_64.

ehmry, 2019-04-09:
With pleasure

closed 2019-04-09 by ehmry