genodelabs/genode #3270: Update depot content implicitly via run script

nfeske, 2019-04-01:
As remarked by @ehmry in https://github.com/genodelabs/genode/issues/3261#issuecomment-478034969, the interplay of the depot with the run scripts can become frustrating because the creation of missing or outdated depot content requires manual steps. To accelerate our work flow, it would be nice if the execution a run script automatically triggered the creation of depot content and update depot-archive versions as needed.

Until now, I was hesitant to implement such a feature into the build system because this would be a violation of the layering of tools: Everything done from within the build directory should stay in the build directory. The build process should - in particular - never modify the source tree.

Well, layering is good and all, but at the end of the day, users don't care about nice layering, but about a smooth operation of the tools.

To make the tools more convenient, I'd like to add a new `--depot-auto-update` option to the run tool. When added to the `RUN_OPT` variable, this option triggers the depot/create operations as needed by the run script. This includes version updates and the build of packages.

nfeske, 2019-04-01:
@ehmry would you like to give https://github.com/nfeske/genode/commit/e18cfd44248d0e59035f24da5f0d3c6bbe488618 a try?

ehmry, 2019-04-01:
Yes, much better.

ehmry, 2019-04-02:
Now it seems everything gets rebuild on every run - `UPDATE_VERSIONS=1 FORCE=1 REBUILD= KEEP_BUILD_DIR=1`.

ehmry, 2019-04-02:
I take it back, just the api and source archives.

commit 1ae77dd1 (2019-04-02): run: add --depot-auto-update feature

If enabled, this option triggers the automated management of depot
content according to the needs of a run script.

Fixes #3270
commit 0ec04c5e (2019-04-03): run: add --depot-auto-update feature

If enabled, this option triggers the automated management of depot
content according to the needs of a run script.

Fixes #3270
commit a3411c8e (2019-04-09): run: add --depot-auto-update feature

If enabled, this option triggers the automated management of depot
content according to the needs of a run script.

Fixes #3270
closed 2019-04-09 by chelmuth