genodelabs/genode #3273: toolchain: update RISC-V to gcc 8.3.0

ssumpf, 2019-04-03:
* build toolchain/binutils for RISC-V ISA version 1.9.1
* make Genode's RISC-V port compile with toolchain
* switch from Spike emulator to Qemu 3.1
* check dynamic linking support (was eager, might be lazy by now)
* check floating point support (issue https://github.com/genodelabs/genode/issues/2481) and maybe implement support in kernel

ssumpf, 2019-04-03:
Since there is no general toolchain update issue I gonna base this on https://github.com/cproc/genode/tree/tool_chain-20190402

ssumpf, 2019-04-25:
In the current state 9cffaed, 78713f8, and  ade069a enable traditional (`run/log`) RISC-V support for GCC 8.3.0.However, the migration from Spike to Qemu introduces the trouble that Qemu only supports "RISC-V User-Level ISA V2.2" while Genode currently implements V2.1 which leads to, yet unknown, issues during kernel context switches. Therefore, I will stick to Spike for the time being until it becomes clear what ISA our updated libc (https://github.com/genodelabs/genode/issues/3289) supports.

chelmuth, 2019-05-29:
Fixed in master.

closed 2019-05-29 by ssumpf