genodelabs/genode #3274: Add SYNC and TRIM as async i/O operations to block session

nfeske, 2019-04-03:
The current block-session interface should be improved in the following respects:

* The `sync` operation is currently an RPC function. It should be an asynchronous operation to ensure the consistent ordering of sync operations with pending read/write requests. This change would also simplify server implementations if they depend on asynchronous I/O for implementing sync. An blocking RPC is bad in this situation because it blocks all other clients until the sync of one client has completed.

* In addition to READ, WRITE, and SYNC, we should add an asynchronous TRIM operation to mark a range of blocks as no longer needed.

* The alignment of the request payload is currently set to 2 KiB by default. As far as I remember, the value was originally chosen to accommodate the ATAPI driver. In general, however, the alignment depends on the server implementation, e.g., it may be dictated by DMA-buffer alignment constraints of a zero-copy block driver. For non-device block servers, the alignment (beyond the block size) should better be avoided to avoid fragmentation of the payload buffer. Consequently, we should introduce a new RPC function `block_alignment` that should be called by the client to request the proper alignment from the server.

nfeske, 2019-04-03:
Until now, each packet of the block session could be uniquely identified by its offset into the payload buffer because each packet - by definition - owns a disjoint part of the packet-stream buffer.

The new operations SYNC and TRIM do not have any payload. Hence, the offset does no longer suffice to uniquely identify packet descriptors. E.g., when a client issues two consecutive SYNC requests, both packet descriptors would look the same. When receiving the acknowledgements, the client cannot know which ack belongs to which request.

For this reason, we should equip the packet descriptor with a client-defined _tag_ value that enables the client to associate requests with their acknowledgements, independent from the other packet-descriptor arguments. The server would preserve the tag as an opaque value.

@m-stein and @cnuke, this topic is related to your current discussions. Some days ago, I argued against adding the `tag` member to the `Block::Request` but now I came to the contrary realization. Please give me a short notice if you are with me.

cnuke, 2019-04-03:
Fine with me.

m-stein, 2019-04-03:
Fine with me too.

commit 13d399d4 (2019-04-10): block_session: SYNC and TRIM as async operations

This patch removes the blocking Block::Session::sync RPC function and
adds the asynchronous operations SYNC and TRIM to the block session's
packet-stream interface.

Even though the patch adjusts all block components to the interface
change, the components keep the former blocking handling of sync
internally for now because of the design of the 'Block::Driver'
interface. This old interface is not worth changing. We should instead
migrate the block servers step by step to the new
'Block::Request_stream' API.

Fixes #3274
commit af4da418 (2019-05-03): block_session: SYNC and TRIM as async operations

This patch removes the blocking Block::Session::sync RPC function and
adds the asynchronous operations SYNC and TRIM to the block session's
packet-stream interface.

Even though the patch adjusts all block components to the interface
change, the components keep the former blocking handling of sync
internally for now because of the design of the 'Block::Driver'
interface. This old interface is not worth changing. We should instead
migrate the block servers step by step to the new
'Block::Request_stream' API.

Fixes #3274
commit 593e9711 (2019-05-06): block_session: SYNC and TRIM as async operations

This patch removes the blocking Block::Session::sync RPC function and
adds the asynchronous operations SYNC and TRIM to the block session's
packet-stream interface.

Even though the patch adjusts all block components to the interface
change, the components keep the former blocking handling of sync
internally for now because of the design of the 'Block::Driver'
interface. This old interface is not worth changing. We should instead
migrate the block servers step by step to the new
'Block::Request_stream' API.

Fixes #3274
closed 2019-05-06 by chelmuth