genodelabs/genode #3275: block_session: remove Block::Session::Operations

nfeske, 2019-04-03:
The `Block::Session::Operations` class was meant for the negotiation of the block session's features between client and server. But some years down the road, it turns out that the only meaningful distinction exists between read-only and read-writeable sessions. Still, all block servers are bothered with the clunky `Operations` definitions.

As a preparatory step for issue #3274, I'm going to replace the `Operations` with a straight-forward `writeable` RPC function.

nfeske, 2019-04-03:
Instead of introducing a new `writeable` RPC function, I'm going to change the current `info` RPC function to return a structured type, which includes the `writeable` attribute.

cnuke, 2019-04-03:
Not that I have specific use-case in mind but maybe it is reasonlable to also add a `readable` attribute (I'm thinking along the lines of a append-only block device). Than again such a attribute could be added when the need arises.

nfeske, 2019-04-03:
In principle, it is easy to add further attributes but I'd wait for the actual use case.

nfeske, 2019-04-03:
Commit https://github.com/nfeske/genode/commit/f3579460851efb69cd497f89f84f0836634a148b implements this idea.

nfeske, 2019-04-04:
Commit https://github.com/nfeske/genode/commit/a11da5d784a72cc2896efd13be31dab6ba237365 presumably covers all block components in the source tree. @cnuke, could you please have a glance?

cnuke, 2019-04-04:
@nfeske thanks for updating the components. So far it looks good. I'm only wondering if we should break the compatibility of Block components immediately without prior notification.

nfeske, 2019-04-05:
@cnuke your remark to be careful about the API change is appreciated.

After sleeping over it, I think that the risk to disrupt Genode users should be acceptable as users rarely deal with the development of block-level components directly. So most breakage happens within our tree where we can take care. In genode-world, there are two components (lwext4 and blk_shred) that I will adjust once the commits enter the master branch.

Nevertheless, once the commits enter the staging branch, I'll post a heads-up message to the mailing list. The adjustment to the API changes is not complicated - rather cosmetic - after all.

The removal of the block/component.h API would be a different story. Here we would need a proper API-deprecation plan.

ehmry, 2019-04-11:
Solo5 fix at 2d9ef64a86fc82eab171cc975794c438b8a01c45.

ehmry, 2019-04-11:
Fixup for solo5 at dfaafa5bbe91c94f79ad03dd35ad6df0437ebc96.

commit c737a0ac (2019-04-09): block session: remove Block::Session::Operations

This patch modernizes the 'Block::Session::info' interface. Instead of
using out parameters, the 'init' RPC function returns a compound 'Info'
object now. The rather complicated 'Operations' struct is replaced by
a 'writeable' attribute in the 'Info' object.

Fixes #3275
commit 2208220c (2019-05-03): block session: remove Block::Session::Operations

This patch modernizes the 'Block::Session::info' interface. Instead of
using out parameters, the 'init' RPC function returns a compound 'Info'
object now. The rather complicated 'Operations' struct is replaced by
a 'writeable' attribute in the 'Info' object.

Fixes #3275
closed 2019-05-06 by chelmuth