genodelabs/genode #3276: VFS block plugin is still synchronous

ehmry, 2019-04-04:
For better or worse the block plugin of the VFS still blocks for packet round-trips to the block service, with implications for VirtualBox and the EXT2 tools.

I would like to convert this make asynchronous I/O and write tests for using the the plugin from Noux, because #3257 shows that there is a shortage of Noux I/O tests.

nfeske, 2019-04-04:
I agree. But since I'm currently busy with reworking the block-session interfaces, let's postpone this idea until I'm finished with #3275 and #3274.

BTW, this is related to #2263. So I'm happy that you identified the VFS block topic as interesting to work on. ;-)

ehmry, 2019-04-04:
I would still go through with a conversion for the sake of testing, but postpone merging until the session changes are finalized.

ehmry, 2019-04-04:
There is already a test.

closed 2019-04-04 by ehmry