genodelabs/genode #3277: hw: nic_router test triggers kernel re-entrance

chelmuth, 2019-04-04:
On imx53qsb and x86_64, we got the following log last night.
```
Kernel: Cpu 0 error: re-entered lock. Kernel exception?!
```

skalk, 2019-04-04:
@chelmuth I will have a look.

skalk, 2019-04-05:
Again, the kernel tried to print using `warning`, but core's console log was held by a core thread at that time. We have to find a general solution here, but for the time-being I've eliminated all direct calls of log/warning/error in the kernel and replaced it by `raw`, which is lock-free.

skalk, 2019-04-05:
@chelmuth please, consider commits 356eab2 and 5130c1f for staging, it fixes this issue for the time being. A solid, general fix for the log/lock problem in base-hw is necessary, but should be part of a separate issue.

commit e1bd99c5 (2019-04-05): hw: replace lock-safe log variants in kernel

As far as possible remove usage of warning/error/log in the kernel,
otherwise the kernel context might try to take a lock hold by a core
thread, which results in a syscall to block.

Fix #3277
closed 2019-04-09 by chelmuth