genodelabs/genode #3280: hw: core's Log console is thread-safe and should not be used by the kernel context

skalk, 2019-04-05:
Lastly issue #3277 again showed this structural problem in base-hw's kernel/core implementation. The background is the following: as we have log/warning/error spread around in Genode's base code it can badly interfer when base-hw's kernel context's code path stripes one of those printing routines. The reason is that they require to take a lock, which can be obtained by a core thread. In that case, the kernel context puts itself into the waiter queue and wants to go to sleep via a syscall.
I see no easy solution to this.

tomga, 2019-04-05:
Was this always manifesting with ``re-entered lock. Kernel exception?!`` or could it be dependent on sequence of events?

Many times I had output truncated on rpi3 when I added too much logs to investigate what happens - sometimes in core (pic and timer drivers). I assumed I have some error in uart, pic or timer which I wrote but maybe I was wrong.

skalk, 2019-04-08:
The concurrent usage of the uart by the kernel and a core context was ending up in previously described error (`re-entered lock`) only. What you might have seen is: core messages that are intercepted by kernel messages. So, if you spread `raw` messages (lock-free) into the kernel code that is executed quite often, it gets impossible to read the other core and user messages. But you should not loose any character.

nfeske, 2021-02-26:
@skalk are there any steps planned to solve this issue?

skalk, 2021-02-26:
@nfeske well no, because it got supressed by daily work items. Anyway it is still a serious problem in base-hw that needs to be fixed in my eyes. So although I cannot say when, we should not forget about it.

closed 2023-01-24 by chelmuth