genodelabs/genode #3281: Libc clock wrap-around

ehmry, 2019-04-05:


ehmry, 2019-04-05:
The `clock_gettime` procedure may be unreliable on 32bit platforms after about an hour due to wrap-around in microsecond values. @al-sharp I believe this is the issue you encountered with QTimer.

Fixed at d5f9a108ad75790331c6d66a60ef0d4d2be46fec and 69413b0ec005729b93bf2193dda62b2761380255.

ehmry, 2019-04-05:
I believe fixup 106d077c8c347b0e9adc5e6d1efbaec2e65b092d is more accurate.

al-sharp, 2019-04-05:
I will try to backport the relevant part of this to 18.08 and see if I can escape reliving the same hour forever.

Should `time_us` be `unsigned long`?

ehmry, 2019-04-05:
It shouldn't matter, 1 second of microseconds should fit in 20 bits.

al-sharp, 2019-04-07:
I tried this patch (in 19.02 staging actually since backport to 18.08 appears to be a bit of effort). Unfortunately the QTimer application still hangs (screenshot of frozen app below). But I do get a meaningful warning I've never seen before: 
```
Warning: libc: limiting exceeding timeout of 4295006 ms to maximum of 4294967 ms
```
Possibly I failed to fix things up per your latest...I made change to duration.*, ld, and timer.cc.

Regarding `time_us` type,  I must be missing something obvious--does not `trunc_to_plain_us()` add `_hours` (expressed in microseconds) to its return value? is it known to be 0?

[image]

chelmuth, 2019-07-09:
Could anybody help me to understand the current state of this issue in the light of our 64-bit timestamp work?

chelmuth, 2019-12-20:
Stale issue.

closed 2019-12-20 by chelmuth