genodelabs/genode #3282: IPv6 nic_router

ehmry, 2019-04-07:
I would like to pass IPv6 packets between Nic sessions. The first scenario I would like to implement is a VM that announces itself as an IPv6 router to other VMs on a common link.

@m-stein do you anticipate the nic_router to support both IPv4 and IPv6? If so, would the nic_router domains support one or both protocols?

chelmuth, 2019-04-08:
Maybe I miss something but your initial scenario sounds like bridging, right?

ehmry, 2019-04-08:
Yes, that's right, first I want the `nic_bridge` to support IPv6 and [Neighbor Discovery Proxies] (https://tools.ietf.org/html/rfc4389). If I can get the `nic_bridge` working then I would open another issue for the `nic_router`, but the `nic_router` needs more planning.

m-stein, 2019-04-08:
I don't know how important these features are. Is there a critical use case currently? I don't think that there are really difficult tasks on the way of integrating IPv6 into the NIC router/bridge. But the configuration interfaces of them may receive non-backwards-compatible changes as the interfaces were designed without considering the existence of another IP version.

ehmry, 2019-04-14:
It seems that the only thing preventing cross-session IPv6 at the nic_bridge is a lack of proper multicast. I'm fine with simply fixing this for now and closing this issue. From what I understand the nic_bridge still needs IPv6 discovery proxying to support IPv6 accross the uplink.

Patch to close this issue at 02e2c1e1d75b822533f82050672a5fdb622deaed.

chelmuth, 2019-05-07:
If I understand correctly with my limited knowledge of IPv6 implementations, traffic no directed to any MAC behind the bridge is directed to ethernet multicast (which includes broadcast) addresses. Is there any source for this information (e.g., an RFC)?

ehmry, 2019-05-07:
Mulitcast packets are used during IPv6 configuration, but once the next hop has been found it should be unicast. I believe the proper RFC is 4861.

In the case of the nic_bridge, the bridge should be intercepting and rewriting the multicast discovery packets as they are forwarded to and returned from upstream.

commit 08cba256 (2019-05-14): Ethernet multicast support at nic_bridge

Fix #3282
commit 4caffd79 (2019-05-27): Ethernet multicast support at nic_bridge

Fix #3282
closed 2019-05-29 by chelmuth