genodelabs/genode #3286: rump ext2fs: crash and corruption

skalk, 2019-04-10:
I've encountered the following rumpfs crash in my sculpt scenario:
```
[runtime -> ahci-1.2.fs] rump: bad directory entry: reclen is too small for name_len
[runtime -> ahci-1.2.fs] rump: offset=364, inode=1581069, rec_len=20, name_len=102 
[runtime -> ahci-1.2.fs] rump: panic: ext2fs_dirbadentry
[runtime -> ahci-1.2.fs] rump: rump kernel halting...
[runtime -> ahci-1.2.fs] rump: halted
[runtime -> ahci-1.2.fs] Warning: rumpuser_sp_fini called (from 0x111b61d) not implemented
[runtime -> ahci-1.2.fs] Error: Rump panic
[runtime -> ahci-1.2.fs] Error: Io_progress_handler subclass cannot be safely destroyed!
[runtime -> ahci-1.2.fs] Warning: 26 dangling allocations at allocator destruction time
[runtime -> ahci-1.2.fs] Error: freeing non-empty slab block
[runtime -> ahci-1.2.fs] Error: ID space not empty at destruction time
[core] no RM attachment (READ pf_addr=0x187e0 pf_ip=0xe2906 from pager_object: pd='init -> runtime -> ahci-1.2.fs' thread='ep') 
[core] page fault, pd='init -> runtime -> ahci-1.2.fs' thread='ep' cpu=0 ip=0xe2906 address=0x187e0 stack pointer=0xa02fef18 qualifiers=0x4 irUwp reason=1
[runtime] child "ahci-1.part_block"
[runtime] child "ahci-1.2.fsck.ext2"
```

Afterwards, I did a filesystem check with quite a lot of issues:
```
[runtime -> ahci-1.2.fsck.ext2] e2fsck 1.42.9 (28-Dec-2013)
[runtime -> ahci-1.2.fsck.ext2] /dev/block contains a file system with errors, check forced.
[runtime -> ahci-1.2.fsck.ext2] Pass 1: Checking inodes, blocks, and sizes
[runtime -> ahci-1.2.fsck.ext2] Deleted inode 4399106 has zero dtime.  Fix? yes

... (more ~20 similar error follow)

[runtime -> ahci-1.2.fsck.ext2] Pass 2: Checking directory structure
[runtime -> ahci-1.2.fsck.ext2] Entry 'DRM.pdf^@pdf' in ??? (1581057) has illegal characters in its name.
[runtime -> ahci-1.2.fsck.ext2] Fix? yes

... (more ~30 similar error follow)

[runtime -> ahci-1.2.fsck.ext2] Pass 3: Checking directory connectivity
[runtime -> ahci-1.2.fsck.ext2] Pass 4: Checking reference counts
[runtime -> ahci-1.2.fsck.ext2] Unattached inode 1581069
[runtime -> ahci-1.2.fsck.ext2] Connect to /lost+found? yes
[runtime -> ahci-1.2.fsck.ext2] 
[runtime -> ahci-1.2.fsck.ext2] Inode 1581069 ref count is 2, should be 1.  Fix? yes
[runtime -> ahci-1.2.fsck.ext2] 

... (more ~5 similar error follow)

[runtime -> ahci-1.2.fsck.ext2] Pass 5: Checking group summary information
[runtime -> ahci-1.2.fsck.ext2] Block bitmap differences:  -(17596944--17597098) -(17826320--17826348) -(17891856--17891916) -(18186768--18187124) -(18187128--18187174) -(18252304--18252353) -(18317840--18319371) -(18383376--18383407) -(18448912--18448950) -(1864
[runtime -> ahci-1.2.fsck.ext2] 5520--18645629) -(18711056--18711081) -(18776592--18776746) -(18809360--18809388) -(18940432--18940492) -(28475920--28475966) -(28475968--28476316) -(28803600--28803649) -(28836368--28837899) -(29131280--29131313) -(29164048--29164
[runtime -> ahci-1.2.fsck.ext2] 086) -(29196872--29196981) -(29229584--29229609)
[runtime -> ahci-1.2.fsck.ext2] Fix? yes
[runtime -> ahci-1.2.fsck.ext2] 
[runtime -> ahci-1.2.fsck.ext2] Free blocks count wrong for group #537 (32099, counted=32254).
[runtime -> ahci-1.2.fsck.ext2] Fix? yes
[runtime -> ahci-1.2.fsck.ext2] 

... (more ~15 similar errors follow)

[runtime -> ahci-1.2.fsck.ext2] Inode bitmap differences:  +1581163 -4399106 -4456450 -4472834 -(4546562--4546563) -4562946 -4579330 -4595714 -4612098 -4661250 -4677634 -4694018 -4702210 -4734978 -(7118850--7118851) -7200770 -7208962 -7282690 -7290882 -7299075 -7
[runtime -> ahci-1.2.fsck.ext2] 307266
[runtime -> ahci-1.2.fsck.ext2] Fix? yes
[runtime -> ahci-1.2.fsck.ext2] 
[runtime -> ahci-1.2.fsck.ext2] Free inodes count wrong for group #193 (8086, counted=8085).
[runtime -> ahci-1.2.fsck.ext2] Fix? yes
[runtime -> ahci-1.2.fsck.ext2] 

... (more ~15 similar errors follow)

[runtime -> ahci-1.2.fsck.ext2] /dev/block: ***** FILE SYSTEM WAS MODIFIED *****
[runtime -> ahci-1.2.fsck.ext2] 
[runtime -> ahci-1.2.fsck.ext2]         6826 inodes used (0.05%, out of 13107200)
[runtime -> ahci-1.2.fsck.ext2]           40 non-contiguous files (0.6%)
[runtime -> ahci-1.2.fsck.ext2]            0 non-contiguous directories (0.0%)
[runtime -> ahci-1.2.fsck.ext2]              # of inodes with ind/dind/tind blocks: 3421/1345/0
[runtime -> ahci-1.2.fsck.ext2]      6159336 blocks used (11.75%, out of 52412416)
[runtime -> ahci-1.2.fsck.ext2]            0 bad blocks
[runtime -> ahci-1.2.fsck.ext2]            1 large file
[runtime -> ahci-1.2.fsck.ext2] 
[runtime -> ahci-1.2.fsck.ext2]         5323 regular files
[runtime -> ahci-1.2.fsck.ext2]         1494 directories
[runtime -> ahci-1.2.fsck.ext2]            0 character device files
[runtime -> ahci-1.2.fsck.ext2]            0 block device files
[runtime -> ahci-1.2.fsck.ext2]            0 fifos
[runtime -> ahci-1.2.fsck.ext2]            0 links
[runtime -> ahci-1.2.fsck.ext2]            0 symbolic links (0 fast symbolic links)
[runtime -> ahci-1.2.fsck.ext2]            0 sockets
[runtime -> ahci-1.2.fsck.ext2] ------------
[runtime -> ahci-1.2.fsck.ext2]         6811 files
[runtime -> ahci-1.2.fsck.ext2] child /bin/fsck.ext2 exited with exit value 1
```

skalk, 2019-04-10:
Interestingly, I cannot remember to interact with the file-system directly by myself, but via VMs and Scult GUI only.

skalk, 2019-04-11:
As it turned out, it immediately happened again when trying to download the file `L4.14.78_1.0.0_ga_images_MX7DSABRESD.zip` (434MB) into my shared folder. Without changing the shared folder's content, I could reproduce the exact same result 2 times. But when trying to reproduce the same by copying an equally named file of ther same name (with zeroes), I could not reproduce it anymore. The shared folder content seemed to have changed, even the original file could not reproduce the observed behaviour.
There is strong indication that the whole issue is related to the combination shared folder and rump, but now I cannot reproduce the faulty behaviour anymore to proof that.

I have now cleaned my shared folder, made `fsck` under Linux (showing lots of orphaned inodes) and keep my filesystem as it is to check whether it will happen again, or was just a chaining of rare unusal circumstances. For the time-being, I'll close the issue, and open it again if I'm able to reproduce it again.

closed 2019-04-11 by skalk

nfeske, 2019-04-11:
Thank you for the update, and for the braveness to keep the existing file system. ;-)