genodelabs/genode #3291: zynq network driver is kernel dependent

skalk, 2019-04-15:
When working on issue #3180 I stumbled across the NIC driver for cadence_gem cards, which is not only dependent on the zynq board (directly uses some system controller), but on some bse-hw kernel syscalls that are not part of Genode's base API. This is a problem when trying to package the driver.

As far as I understand the code correctly it uses normal dataspaces as RX and TX buffer and uses the kernel API to write back and invalidate the cache before doing DMA. From my understanding it would be better to use uncached dataspaces for those buffers instead, thereby eliminating the need to flush the cache at all. This is the way, we normally write device drivers. But maybe I oversee something relevant. The problem is, I do not have real hardware for this platform - only qemu which doesn' emulate caching. @ValiValpas can you share more insights regarding this?

skalk, 2019-04-15:
I've seen that the driver uses the generic Nic::session_component, which implicitly allocates cached RAM for TX and RX buffers. As solution I've added a cache_policy parameter to the constructor of this class. @nfeske as original author of this abstraction, do you agree on this change (reference commit fa45d9080ec0522be88f9a5ce92ab96192f82f65)
And if so, @ValiValpas would you try the commit whether real hardware still works appropriatedly?

nfeske, 2019-04-15:
I'm fine with the cache-attribute argument. But would you agree to split off this interface change to a commit separate from the Zynq-specific change?

skalk, 2019-04-15:
Sure.

ValiValpas, 2019-04-15:
@skalk I get your point and I'm happy to provide more insights.

I'm afraid this may be in conflict with the zero-copy implementation of the driver. The zynq nic driver directly passes the packets from its client to the DMA controller and vice versa. Packet data that resides in cache must thus be written back before handing it over to the DMA. This is not an issue if the driver copies the packet to uncached memory. However, with the zero-copy implementation this is kind of tricky as the memory that will be accessed via DMA is mapped in multiple virtual address spaces.

I haven't thought this through yet, but is it ensured that the packet buffers are also uncached by the nic client?

skalk, 2019-04-16:
@ValiValpas yes, with above commit the communication buffers used by this driver are uncached no matter where they are mapped to. The cache attributes are always assigned to the dataspace not the individual mapping to circumvent mappings with different cache attributes but same physical area, which is a problem for physically indexed caches. It should work, but as I said I'm unable to test it on hardware.

ValiValpas, 2019-04-16:
@skalk sure. Is it okay, if I test this next week? I'm currently at a conference from where it is rather inconvenient to do this.

skalk, 2019-04-17:
Of course, I only wanted to hint you to not break silently your platforms in use.

nfeske, 2019-04-17:
I merged both commits to staging. We won't update the master branch this or next week. @ValiValpas So testing it next week would be perfectly fine. ;-)

ValiValpas, 2019-04-30:
@skalk I tried running the staging branch on the zynq hardware. Unfortunately, there is a failed assertion in `page_table.h`:

>Starting kernel ...
>
>
>kernel initialized
>Error: Assertion failed: !Descriptor::valid(_entries[i]) || _entries[i] == e
>Error:   File: /home/johanness/repos/genode/repos/base/../base->hw/src/include/hw/spec/arm/page_table.h:234
>Error:   Function: void >Hw::Page_table::Page_table_level_2::insert_translation(Genode::addr_t, Genode::addr_t, >Genode::size_t, const Hw::Page_flags&)

I do not get this error when I execute this with qemu though. I can create a separate issue for this if you like.

I'll try cherry picking your commits to test the nic driver changes in isolation.

skalk, 2019-04-30:
Ok, so things seems to have been broken some time before?! This does not seem to be related to my commit, but already fails in bootstrap.
Thanks for testing the platform, I think I cannot do anything in a separate issue without your engagement ;-). If you like you could add something like `log("add mapping ", m);` at the beginning of the `void Platform::Pd::map(Mapping m)` function inside bootstrap `repos/base-hw/src/bootstrap/platform.cc`. Thereby, we could see which mapping is responsible for the assertion. But actually, I first and foremost did not want to break things for you and your group. So just inspect the issue if it is relevant to you.

ValiValpas, 2019-04-30:
No worries. I haven't used the platfrom for a while so I'll try to bisect what changes may have caused this.

Here is the log output you suggested, what do you think?

> add mapping physical region(base=0x4000 size=0x4000) => virtual address=0xc0000000 with page-flags: writeable, exec, privileged, global, cached)
>add mapping physical region(base=0x2dc000 size=0x101000) => virtual address=0xc0004000 with page-flags: writeable, exec, privileged, global, cached)
>add mapping physical region(base=0x100000 size=0x1d9000) => virtual address=0x100000 with page-flags: writeable, exec, privileged, global, cached)
>add mapping physical region(base=0xf8f00000 size=0x2000) => virtual address=0xd0000000 with page-flags: writeable, noexec, privileged, global, iomem, uncached)
>add mapping physical region(base=0xe0001000 size=0x1000) => virtual address=0xd0003000 with page-flags: writeable, noexec, privileged, global, iomem, uncached)
>add mapping physical region(base=0xf8f02000 size=0x1000) => virtual address=0xd0005000 with page-flags: writeable, noexec, privileged, global, iomem, uncached)
>add mapping physical region(base=0x149000 size=0xa4000) => virtual address=0x80000000 with page-flags: readonly, exec, privileged, global, cached)
>add mapping physical region(base=0x149000 size=0x1000) => virtual address=0xffff0000 with page-flags: readonly, exec, privileged, global, cached)
>add mapping physical region(base=0x8000 size=0x5d000) => virtual address=0x800a4000 with page-flags: writeable, noexec, privileged, global, cached)
>add mapping physical region(base=0x2d9000 size=0x1000) => virtual address=0xfffe0000 with pag�+�[C������
                                                                                                        )�+�+++K��mz�5)��5�j�ɽ��
         �͕�ѥ����������
                      "�͍ɥ�ѽ��鲅����~���ɥ��m�u����}���ɥ��m�u���*mm�5)��5�j�ɽ��2����zB������������ͽɕ�ͽ�������ɕ�ͽ��͕��rz�͕��ݽ�ɍ�����Ց���ݽ�����ɵ����}х������ɚ�ڴj5)��5�j�ɽ��2չ�ѥ��遲����!��邅��}х����邅��}х���}��ٕ�}���J�͕��}�Ʌ�ͱ�ѥ����������
                                                                                                             ���}ѱ�:�������
    ���}ѱ�:������隥�}ѱ����с!��邅��}����͙���j5R�

ValiValpas, 2019-04-30:
@skalk the commit that breaks the zynq platform is 8c460b3ea5a8745b3b7b336b4095f9fac13cd763

I'll have a look at how to initialize the L2 cache for the zynq boards correctly. Let me know if you have any hints.

ValiValpas, 2019-04-30:
I tested your patches for the `Nic::Session_component` and the zynq nic driver with an older Genode version, works like a charm ;-)

There is just a minor issue: I think you missed adding the Genode::CACHED argument to the linux nic driver.

ValiValpas, 2019-04-30:
fyi: I think I found the correct initialisation parameters for the zynq's L2 cache. I'll run some further tests and push the changes to genode-world in the next days.

skalk, 2019-05-02:
@ValiValpas thanks for noticing the Linux driver breaking! @chelmuth already recognized the missing Linux driver and provided a fixup commit 179f38d4d85f41489511373776705a81c729d8ad. It is very cool that you found a fix for the L2 cache initialization!

commit 306b32d4 (2019-04-17): zynq: make nic driver independent of hw kernel

Use uncached ram dataspace for RX and TX communication buffers
instead of cache maintainance functions that are not part of the
Genode API.

Fix #3291
closed 2019-05-06 by chelmuth