genodelabs/genode #3294: vfs_trace: enable tracing support via a VFS plugin

ssumpf, 2019-04-16:
This line of work will replace the aged `trace_fs` file system with a current VFS plugin. My progress can be found under: https://github.com/ssumpf/genode/tree/vfs_trace So far, generating XML for `Directory_file_systems` from `Trace::Session_label`'s is working, also I revived @cnuke's `Value_file_system` class which is now writeable, this way the plugin can react to events. In the next step I gonna implement a `Single_file_system` which gathers trace buffer data. I'll post updates to this thread as soon as a goal is reached.

ssumpf, 2019-04-18:
* added `Watch_handle` support to `Value_file_system`
* removed signals from `Value_file_system`
* enabled `Watch_handler<>` for plugin

ssumpf, 2019-05-02:
* added `Trace_buffer_file_system` as `Single_file_system`
* connected and integrated trace session

ssumpf, 2019-06-13:
Current state update:
* basic functionality - *working*
* `tail -f` support - *working*
* trace buffer resize (also at runtime) - *working*
* start/stop/resume tracing - *working*
* configuration of memory for trace session in `<vfs>`-node - *missing*
* README - *missing*

nfeske, 2019-06-14:
Thanks for sharing the progress. This looks like all the hard parts are solved - congrats!

What do you think about adding a noux-based Sculpt runtime to the list? This would allow everyone to experiment with the new feature live on Sculpt. I know I'd love to. ;-)

ssumpf, 2019-06-14:
@nfeske, @chelmuth: Please review the commits on my https://github.com/ssumpf/genode/commits/vfs_trace_staging branch. I am open for any suggestions.

ssumpf, 2019-06-14:
@nfeske: I have updated the state above by adding the noux runtime.

nfeske, 2019-06-14:
The changes in base look very good to me! I especially like the implementation of the `trace()` function. In the back of my head I was a bit worried that you may pull the `base/thread.h` header into the `log.h` header. It is very good that your commit hides the trace-buffer interaction completely inside the cc file.

I have only glanced quickly over the actual vfs_trace commit so far and like it. It is very good that you separated the re-usable parts (trace buffer, directory tree, and value file system) so well from the trace-specific code. So we can consider moving those utilities to `os` in the future.

In the code, your passing of the `Constructible<Trace_buffer> &` as argument caught my eye. I think it would be more clear to pass a reference to a new interface called `Trace_entries` to the `Vfs_handle`. The implementation of this interface could do the `constructed()` check outside of the `Vfs_handle` class. This way, the logic of the `Vfs_handle` becomes separated from the concern of whether the buffer exists or not. But that's just a suggestion not a big concern of mine. I'm perfectly fine with the code as is.

ssumpf, 2019-06-14:
@nfeske: I have implemented your suggestion and removed the artefacts. So please consider  a8a61c1599d985d12188f841003a1a3189d04115 1b38fc0c4a38a6965a6cef35729381c1edd690e6 b020e033aec44645ecc259ce12ed33833fe0a38b 06c4f7e52d2c5198d35ad46762368650f68896db d371f1b71a9c6f457d3c874e99071282da0535dd for staging.

ssumpf, 2019-06-14:
6db1a0f adds effective c++ conformity

chelmuth, 2019-06-17:
I reviewed the series and merged the commits to staging.

nfeske, 2019-06-17:
> I have implemented your suggestion and removed the artefacts.

That's all the better. Thanks for considering my remarks.

ssumpf, 2019-06-17:
6781f6c adds the missing `vfs_trace.mk` file. 2a494f0 and 24ac528 add the `trace_fs` runtime.

cnuke, 2019-06-18:
With the `vfs_trace` plugin in place I believe it is time to retired the old `trace_fs` component.

nfeske, 2019-11-19:
Fixed in master.

closed 2019-11-19 by nfeske