genodelabs/genode #3295: platform_info: add information about running kernel

skalk, 2019-04-17:
There are few use-cases where it would be good to know on which kernel a component gets executed. As an example we discussed extending runtime test components like depot_autopilot for instance to run certain tests limited to specific kernels. When working on issue #3180 and #2190 I stumbled across the gpio driver for raspberry pi, which has a kernel-specific special case regarding the interrupt number, because Fiasco.OC and base-hw do enumerate on that platform differently.
It would be nice, if we can read the information about the underlying kernel from the already existent platform_info ROM distributed by core.

tomga, 2019-04-17:
@skalk, do I correctly assume that you want to fix gpio driver issue on rpi with some condition in the driver code: ``if <foc> irq=x else if <hw> irq=y``?

If so, I think it would be better to pass configuration to the driver instead. Would it be possible to add
``<gpio irq="..." mmio_base="..." mmio_size="..."/>``
or some other structure with this information?

I ask because I think this work touches broader topic I wanted to discuss during Hack'n'Hike. And it is not hard to guess it is influenced/poisoned by what I know about device tree used in linux (and not only linux).

Generally I think that drivers should receive irq number and other resource information from platform driver. It would allow to use the same driver on different arm boards without any source code changes. Currently for rpi3 source code changes to all drivers are needed even if the driver is identical otherwise because base address for peripherals is different. I planned to implement passing configurations to drivers in some form later when integrating my changes with but I'm not that far yet.

skalk, 2019-04-18:
@tomga in general I absolutely agree that we need configuration for io resources for driver instances. I've already went into this direction in more recent driver developments like FEC driver for freescale NIC devices (reference `repos/dde_linux`). But in this concrete case it is more about different enumerations of interrupts on that specific platform.
In this example, the I/O resources of a concrete device are given - nowadays statically within the codebase for historical reasons - but both kernels count differently. The base-hw kernel takes the absolute interrupt number used at the GIC, whereas Fiasco.OC starts to count from the shared peripheral interrupts only. I think it is ok to hide that kernel detail in the code instead of having the whole matrix of board + device + kernel in form of package descriptions but board + device only.

skalk, 2019-04-18:
@tomga in this commit you can see that the kernel information is only used to add an offset, not to provide the interrupt number in general: https://github.com/skalk/genode/commit/03138b9f8ecc3175666b65967b5dff84a9e9fd94#diff-6e02dcdaee550f0923c5a3f7c9423b6a

tomga, 2019-04-18:
This is a nice commit - it removes something ugly in run scripts in many places - and definitely in good direction in my opinion. But it doesn't solve two things I think would be good to have solved:

1. For rpi3bplus I'll need to compile rpi3bplus_gpio_drv as far as I know only due to different mmio base address.
2. I think driver's should not be able to ask for any resource - it should be validated somewhere that driver needs some resource - irq in this case. This topic is another thing I wanted to discuss when we meet but it's more for the future.

Anyway this commit gives me example how to pass information from platform driver to concrete driver and I'll be able to try to implement my ideas on my own

skalk, 2019-04-18:
As I said, I agree that it would be better to not compile the resource addresses into the device driver code, but make them configureable. Moreover, as you also stated it would be necessary to check the resource constrains in form of a policy. The natural way to do so, would be a platform driver analoque to our x86 equivalent. But until now, we were not at the point to implement a long-term solution fitting all different ARM platforms. We first had to collect enough experiences using a bunch of different platforms, and for that time it was sufficient to have statically compiled resource information in the driver. I would be happy to start a discussion about design and interaction of drivers and platform_driver in the ARM world in a separate issue, to finally substantiate a design.

tomga, 2019-04-18:
@skalk will you create such issue? I didn't feel comfortable with starting such a general issue about a design as I still have only partial view on the system but if you'd like me to do it I can try.

nfeske, 2019-04-18:
@tomga Don't be shy! ;-)

tomga, 2019-04-18:
Well, some time ago I received a comment about "striving for the stars" :-) I didn't stop after receiving it but I really try to not say anything that could be seen as imposing my view. Especially in design and especially when I can't devote as much time for implementation as is really needed.

But summarizing, I'll write my thoughts. Not promising today, but I will.

skalk, 2019-04-18:
@tomga I think you can perfectly do this, but I'll open it anyway, because I've further issues regarding the state of the platform_driver under ARM.

tomga, 2019-04-19:
@skalk thanks. Your version gave me some more information to think about. My version would be more like a wishlist. Your is more like current problems + solution proposal which is much more concrete and closer to implementation. I'll rethink my view and comment there later.

skalk, 2019-04-23:
@cnuke would you mind to take fixup commit 13bdfe6  to your staging for testing purposes?

cnuke, 2019-04-23:
@skalk done.

commit bb9ca56d (2019-04-18): platform_info: add kernel information (fix #3295)
commit 208040dd (2019-04-29): fixup "platform_info: add kernel information (fix #3295)"
closed 2019-05-06 by chelmuth