genodelabs/genode #3296: sculpt: '+' menu resize issue

cproc, 2019-04-17:
Since the commit for issue #3241, when selecting a component from the depot to start, the selected menu entry is not resized immediately, but only after hovering over a menu element, for example the 'Install' button.

nfeske, 2019-04-18:
I can reproduce this issue.

However, it is not related to 01cff3ce8481625f251b7e9dc1549cdfb9c5911b. This would have really surprised me because the commit does not touch the menu view. The sculpt manager (which is affected by the commit) seems to be doing its job just fine because the widgets added to the new dialog version appear. There is also no busy loop in the system, which could - in principle - explain a hypothetical starvation of the menu view.

The problem must be somewhere inside the menu view. I'll investigate,

nfeske, 2019-04-18:
I have synthesized the problem in https://github.com/nfeske/genode/commit/de282c9ca2eec4761e2a9edc73dcb82656df7363  and can now reproduce it on base-linux via the menu_view.run script.

nfeske, 2019-04-18:
Fixed by https://github.com/genodelabs/genode/commit/8c9ed5efbe73adb61ed52b3cf28cdfe3931d1a27, which I just merged to staging.

cproc, 2019-04-18:
I can confirm that this fixes the problem on sculpt as well.

commit 8c9ed5ef (2019-04-18): menu_view: midway re-targeting of geometry motion

This patch fixes the corner case where an animated geometry changes its
destination mid-way while an animation is already in progress. The
'_trigger_animated_geometry' method used to back out early in this case,
which was intended as an optimization.

Fixes #3296
closed 2019-05-06 by chelmuth