genodelabs/genode #3297: ACME client

cnuke, 2019-04-18:
So far, we rely on manually updating the certificate for appliances like *genodians.org* (it is scheduled for next week as the current one will expire on the 25th of April). Ideally the updating process should be automated which requires running a component in the appliance that handles
the ACME protocol as well as a way to handle [domain validation](https://letsencrypt.org/how-it-works/). Since we already run lighttpd in the *genodians.org* appliance and is currently our only use-case, using the HTTP _well-known URI_ approach sounds reasonable.

On the protocol-handling side we could either write (does anybody knows how “involved” the protocol is?) or port an existing client. The `let's encrypt` website lists a bunch of available [clients](https://letsencrypt.org/docs/client-options/) out of which the C, C++ or Python based ones are probably the more easier ones to port. There is also a popular shell script [acme.sh](https://github.com/Neilpang/acme.sh/blob/master/acme.sh) but since we have removed networking support from Noux and a libc component cannot `fork()` it is out of the question for now.

On the domain-validation side we _just_ need to configure lighttpd appropriately to serve the right resource. How the resource gets there might depend on the ACME client - using the file system as transport mechanism would be one way.

Apart from that we have to decide how to handle the information flow or rather how the updating should be handled, i.e., do we bake the private ACME key into the appliance image like we currently do with the certificates and request a new certificate every time the appliance gets started?

So, to break it down, we have to

- investigate and evaluate ACME client options
- look how to configure lighttpd properly
- integrate a client 
- implement the updating process on-target

nfeske, 2019-04-18:
Thank you for the interesting write-up.

cnuke, 2019-04-23:
I updated the certificate (expires on the 22th of July) and will deploy it soon. That buys us some time to tackle this issue.

cnuke, 2019-07-19:
I updated the certificate (expires on the 18th of October). That will buys us some more time to tackle this issue.