genodelabs/genode #3299: platform_drv for ARM universe

skalk, 2019-04-18:
The current state of ARM device drivers for Genode is a diverse conglomeration of approaches. Most of them reference resources from board-specific definitions in `repos/base/include/drivers/defs`, although they are not necessarily bound to a specific board and its resource names. A good example here is the NXP i.MX series we broadly support. Those boards share more or less the same devices. There are even some driver examples in our codebase that depend on a board and a kernel (drivers for rpi).

In more recent drivers we tried to address this problem of re-instantiating the same driver component using different boards resp. resources, e.g., the FEC nic driver. The problem here is that the configuration of the driver is not necessarily enforced when it directly routes to the IRQ and MMIO services of core. Because core implements no policy. Therefore, it would be nice if the drivers get their resources from a middle-component like the `platform_drv` for x86.

Some Genode ARM drivers already depend on a `platform_drv`, but sadly the API for the `platform_drv` is not generic not even with regard to the same architecture. Nowadays, there are `platform_drv` implementations for: arndale  imx53  odroid_x2  rpi  x86, and each of them provide their own interfaces. Things get even more complicated if you prepare packages for those device and platform drivers, because the depot packages are principally build for an architecture and not a specific board, e.g., refer to the existent platform_drv package.

My proposal for a step out of chaos, would be the following:

Move the current x86 `platform_session` API to be a generic one. It actually aggregates the possibility to iterate over devices and to get DMA capable memory. Both aspects would be useful for ARM too. The `platform_device` on the other hand will stay architecture dependent (although the current x86 variant is more relying on PCI bus properties, which can get interesting for ARM too). For ARM we need to get the I/O memory mapped registers in form of a dataspace here, as well as the interrupt resource. Moreover, we need the ability to turn the device on and off. For the time being I think it would be ok to implicitly do this when claim/release the device.
Then we have a zoo of different information a driver might be interested in, like clock rates, framebuffer dimensions (from the rpi gpu), MII bus configurations, and so on. Because this would blow up a sane API, we need a generic way of exporting that configuration to the driver. A config rom exported by the `platform_device` naturally comes in my mind.

tomga, 2019-04-25:
@skalk could you plese answer my questions below? That would help me to verify if correctly understand current situation.

> Move the current x86 platform_session API to be a generic one. It actually aggregates the possibility to iterate over devices and to get DMA capable memory. Both aspects would be useful for ARM too. 

My feeling is that iterating over devices shouldn't be needed (unless there are cases when one instance of a driver needs to handle more than one device instead of many distinct driver instances). Device driver should declare what devices it is capable to handle and in return it should be given a device.
Comments in #1359 suggest that in 2015 it was a goal for some point in the future. Did it change in the meantime or just not implemented yet?

> The platform_device on the other hand will stay architecture dependent (although the current x86 variant is more relying on PCI bus properties, which can get interesting for ARM too). For ARM we [...]
Then we have a zoo of different information a driver might be interested in, like clock rates, framebuffer dimensions (from the rpi gpu), MII bus configurations, and so on. Because this would blow up a sane API, we need a generic way of exporting that configuration to the driver. A config rom exported by the platform_device naturally comes in my mind.

Do I correctly understand that after implementing passing miscellaneous information to drivers using config rom on arm the same technique could be used to pass information to current drivers on x86? It seems to me that most of current platform device api specific to pci could be replaced with it. Am I right?

It would allow to have one api for different devices by hiding bus/device specific information in config rom. I'm not sure such change would be a good but passing information as xml is a common practice in genode.

skalk, 2019-04-26:
@tomga I try to answer your questions, regarding this discussion starting point. It is not written in stone. I'm glad about all constructive critics, like yours.

> My feeling is that iterating over devices shouldn't be needed (unless there are cases when one
> instance of a driver needs to handle more than one device instead of many distinct driver instances). [...]

The iteration is meant exactly for the rare case of having multiple devices handled by one driver. Imagine several identical USB host controllers or GPIO controllers, where it is not interesting to differentiate in between.
It is not meant here that the driver sees other devices than the once used by it, which are written into the policy of the platform_drv. The driver only iterates over *dedicated* devices. In the common case a device driver will only see one device, and might be implemented in a way that it takes the very first device without caring about iteration at all.

> Do I correctly understand that after implementing passing miscellaneous information to drivers using
> config rom on arm the same technique could be used to pass information to current drivers on [...]

I think it would be possible to put certain PCI related information into such a ROM, e.g., bus addresses, but there are some more x86 and PCI specific functions in the current platform_device abstraction, like I/O port allocation and writing to PCI config space, that do not apply to this scheme. Anyway, in the first steps I would be glad to have a common ARM platform_device API. Having an architecture-independent API would be something several steps beyond.

trimpim, 2019-04-29:
@skalk would platform_drv also manage which driver gets access to an Irq_session? I'm asking, because on sel4 it isn't possible to return an Irq_session to the kernel.
This leads to a problem in the sd_card_drv, as it requests an Irq_session when a client connects and returns it when the client disconnects.

alex-ab, 2019-04-29:
> @skalk would platform_drv also manage which driver gets access to an Irq_session? 

Yes, like on the x86 version already done.

> I'm asking, because on sel4 it isn't possible to return an Irq_session to the kernel.
> This leads to a problem in the sd_card_drv, as it requests an Irq_session when a client connects and returns it when the client disconnects

Please be careful or be better precise. The red message tells you that freeing is not implemented by Genode, it does not necessarily means the kernel does not support that. I think it is just a missing feature one should implement if the one needs it ;-)

trimpim, 2019-04-30:
@alex-ab correct, the message just says that it isn't implemented in Genode.

I already have spent about a half day trying to understand how I could implement it. Unfortunately I wasn't able  to see how this would have to be done from the sel4 documentation.

As I can use base-hw and have more pressing things to do I can't investigate further and let it be as it is at the moment.

skalk, 2020-04-15:
Ok, I'd like to pick up this issue again. Maybe @tomga and @tworaz are interested in the discussion too.
To be in line with the roadmap 2020 (20.05), I would like to go the first steps to have a common ARM platform driver, like described in my first message of thiss issue a year ago. That means:
* Move Platform_session from x86 specific include path to generic place
* Introduce first simple ARM-specific but SoC-common Platform_device
* Replace RPI and i.MX53 platform driver with the new common API
* Adapt the drivers for RPI and i.MX53 to use the new driver

For the time-being I would like to reduce the Platform_device to only provide I/O memory and the IRQ capability (and DMA memory). The device would get powered implicitely when the Platform_device gets requested, and powered off when closed again.
In a later step, we might add clock rate get/set to the interface. By now, I cannot see any driver that makes use of it.

Currently, there exists a `board_revision` method in the platform_session API of the i.MX53, which can be moved out perfectly by using the platform_info being available as ROM module in the meanwhile.

The only open question to me is the `setup_framebuffer` method of the Raspberry Pi's platform driver. I'm not sure where to move this resp. how to change this. To me it seems necessary to keep that code inside the platform_driver, because the so-called 'Mailbox' mechanism used to set/get the framebuffer resolution is also used for power states of the devices. It cannot be split, or safely exported to the framebuffer driver itself. Therefore, we need a way to preserve the possibility to handle the framebuffer. The open question is how to preserve that functionality, but get a sane, common Platform_device interface without these kind of extensions. Maybe @nfeske or @chelmuth have an opinion here too.

tomga, 2020-04-15:
@skalk I'll comment on two things.

**`setup_framebuffer`**
Your proposal mentions three APIs:
 - new platform driver API for handling devices - of course implemented in platform driver
 - power API for enabling for powering on and off devices, that you do not want to expose now as a service and plan to implement inside platform driver
 - some framebuffer setup interface that currently is part of rpi platform driver currently (but probably should not)

Given that second and third have to be implemented in one component due to using one physical device and you want (at least for some time) first and second implemented in one component then you have to implement everything in one component.

If the `setup_framebuffer` was moved to a new API it seems to be easy to later move implementation of it to a new component which will handle powering on/off devices. This issue is definitely not about support for multiple monitors (I briefly remember someone mentioning it) but I think this currently rpi specific `setup_framebuffer` probably could become a part of such API. When I think what options are available using `xrandr` on linux and `setup_framebuffer` they seem to cover similar functionality.

**DMA**
Currently there is no (as far as i know) clean solution for translating bus addresses to physical addresses. It would be great if this was somehow resolved before or during implementing this issue. Currently I had to add ugly hacks to perform those translations in different drivers. It would be great if this was done by the platform driver and passed to drivers. Possibly you remembered about it but just in case I bring it here.

chelmuth, 2020-04-16:
@skalk I like your plan for the first big step. Please do not hesitate to propose adaptions (and future improvements) of the x86 platform driver during the unification. Regarding your question...

> The only open question to me is the `setup_framebuffer()` method of the Raspberry Pi's platform driver.

I doubt we will be able to completely shun such platform specifics from the platform driver and its interface. If anything, I'd expect the TEE functionality yielding even more surprises when we complete our platform support in the future. Nevertheless, I have no universal formula but would suggest to follow the direction of `Native_pd` and `Native_cpu` and provide a hook for board-specific device interfaces.

https://github.com/genodelabs/genode/blob/46fa8197a21289f54bb96c814bf62fc14d149c6e/repos/base/include/pd_session/pd_session.h#L295-L298
https://github.com/genodelabs/genode/blob/46fa8197a21289f54bb96c814bf62fc14d149c6e/repos/base-nova/include/nova_native_pd/nova_native_pd.h#L22-L23

@tomga for DMA-buffer allocation I expect the ARM platform driver inherits the following features from the x86 implementation.

https://github.com/genodelabs/genode/blob/46fa8197a21289f54bb96c814bf62fc14d149c6e/repos/os/include/spec/x86/platform_session/platform_session.h#L75-L83

skalk, 2020-04-16:
@tomga thank you for participating in the discussion.

> Your proposal mentions three APIs:
> 
>  *   new platform driver API for handling devices - of course implemented in platform driver

Not new but the existing one from x86 called `Platform_session`, but yeah you're right.

>  *   power API for enabling for powering on and off devices, that you do not want to expose now as a service and plan to implement inside platform driver

For the time being, I would like to hide this, and instead power up/down devices when request/release device resources, to facilitate the next step of unifying `Platform_device` API in between the architectures. When we have a use-case for dynamic device enabling/disabling maybe with PCI devices and native ARM platform device, we might make powering explicit.
By now, we have no device related power-management anywhere in our stack.

> *   some framebuffer setup interface that currently is part of rpi platform driver currently (but probably should not)
>
> Given that second and third have to be implemented in one component due to [...]

true.

@chelmuth thank you for your thoughts too!

> I doubt we will be able to completely shun such platform specifics from the platform driver and its interface. If anything, I'd expect the TEE functionality yielding even more surprises when we [...]

After dithering about it yesterday, a short discussion with @nfeske about it, and reading your comment, I think you are right. Especially, the way to encapsulate it cleanly in a `Native_device` like manner seems to be the best approach. On the one hand it prevents from having fragmented `Platform_device` APIs, otherwise there is no need for additional routing to another session API, because the capability comes with either the `Platform_session` or `Platform_device`. Thanks, a lot I see now the horizon.

Another point, I've discussed offline yesterday is the board revision information. Moving it to the `platform_info` ROM does not seem to be the best solution on the second view. The "board" information is primarily a SPEC variable right now that hints the right include path to build kernel/core e.g. for base-hw. We would need a bunch of new headers for each board, and include pathes in the generic base directory, and everything because of one driver (i.MX53 framebuffer), which distinguishes in between tablet and Quickstart board. It is much better to add a corresponding configuration value to the driver, and not to bloat the `platform_info` this time.

skalk, 2020-04-16:
Last but not least, I've thought about a device-tree solution for long now. After feeding back this with @nfeske too, my current plan is to provide a one-to-one translation of the device-tree (provided from outside) into a XML configuration compliant with Genode's XML parser. Thereby, we do not have to add additional parsing routines in a low-level component like the platform driver. Moreover, in the future it will be possible to export device-specific parts of the tree inside the `Platform_device` API, e.g. in form of a ROM. When using XML here, we can recycle also the XML generator, and native Genode drivers do not need to read another syntax (device tree notation) here too.
On the other hand to lower friction of maintaining XML-like device trees it must be easy to automatically create them out of existing Device Tree Binaries, either by offline tools, or by a report component analogue to the ACPI report delivered to the platform driver on x86.
The platform driver itself, just interprets parts of the tree, like bus-addresses of devices to identify the right MMIO dataspaces, or interrupts.

In SoC specific implementations, other information like clocking, powering, and pinning can be consumed too. But that would be a second step.

An interesting side aspect would be how to transparently hand over IRQ vs. GPIO interrupts of "platform devices" to drivers. Or whether we can unify I/O port accesses and device register accesses via I2C or SPI or other buses, and handle this transparently too. But those thoughts are not fully perfected, and we can discuss it when the first steps are made.

Anyway, if you like to add something to the topic of device tree and platform driver consolidation, now it's the best time to do so.

tworaz, 2020-04-16:
If you aim to translate device trees into XML I'd be strongly in favour of doing it in runtime. At least in case of Qemu virt platforms the DTB binary is constructed dynamically to reflect the emulator configuration. Passing different command line switches or changing their order does affect the contents of the DTB blob in a non trivial way (different irq numbers, device addresses etc) .

skalk, 2020-04-16:
@tworaz I can see your argument. Well, there is no limitation in doing so in general. A component like the one presenting an ACPI ROM to the platform driver on x86 could do so as well.

But I think it is not crucial to do this in the first place. When using the run-tool to start scenarios on Qemu, we can use always the same arguments anyway. Your corresponding pull-request for network scenarios goes in the right direction here in my eyes by always returning the same parameters to retain the same peripherals. The only thing that changes in between different run-scripts apart from the peripherals, is the size of RAM. The memory size however is hardcoded inside the kernel nowadays anyway. The kernel does not consume any device tree information. Therefore, I would leave this dynamic transcoding open for the time-being.

Of course, the offline tooling has to be easy to apply, so that everybody who wants to change something, or wants to use any DTB out there in the wild, can do so.

skalk, 2020-05-22:
@chelmuth @nfeske for the time-being, I have finalized a first version of a common ARM platform driver, and adapted the driver-nic and drivers-interactive for pbxa9 to use it. All needed commits are on top of my merge_to_staging branch and referenced above in this issue. In my eyes this is a good state to publish within the release, as long as you agree.

nfeske, 2020-05-24:
Even though I haven't reviewed the commits yet, I merged them to staging for the automated testing just now. I'm looking forward to taking a close look at the patches tomorrow.

nfeske, 2020-05-25:
Such a beautiful commit series!

My few suggestion are as follows:

* The `Connection::xml()` method should better become a `Connection::with_xml(FN const &fn)` method, which avoids the passing of `Xml_node` as return value, improving memory safety. In the longer term, `Xml_node` may become `Noncopyable`. If we can avoid copying `Xml_node` objects now, we can thank ourselves later. :-)

* DMA buffers could be referenced using a session-local ID space instead of taking the `Ram_dataspace_capability` as key. This takes the capability-reidentification problem (of several kernels) off the critical path.

* You may consider importing the `Genode` namespace into the `Platform` namespace because the latter is tightly coupled to the former anyway. This would avoid many repetitive `Genode::` prefixes, improving - in my opinion - the clarity of the code.

Those suggestions are not critical for the release though.

skalk, 2020-05-25:
@nfeske thanks you for the compliment! I've put a fixup commit series on my merge_to_staging branch, which:
* fixes the arm_v6 compilation issue
* turns the `List<Session_component>` into `Registry<Session_component>`
* includes `Genode` namespace into `Platform` namespace in the session API
* turns `Connection::xml()` into `Connection::with_xml(FN const &fn)`
I will rethink the DMA buffer API to use session-local IDs instead of the `Ram_dataspace_capability` soon as well, but maybe it is too touchy for the release.
Anyway, thanks for the careful review!

nfeske, 2020-05-25:
Wow, that was quick! Thank you. I added your fixup commits to staging.

nfeske, 2020-05-25:
> includes Genode namespace into Platform namespace in the session API

BTW, this may also allow for the (later?) simplification of the platform driver's implementation.

nfeske, 2020-07-03:
Given the fact that the new platform driver for ARM was featured by the Genode 20.05 release, may we close this issue?

@skalk I think the subsequent platform-driver-related topics deserve separate issues. What is your opinion?

skalk, 2020-07-06:
@nfeske: I agree

closed 2020-07-06 by skalk