genodelabs/genode #3301: VFS plugin is unintuitive

ehmry, 2019-04-20:
The VFS audit plugin would be more intuitive if it would host sub-file-systems within itself, like so
```xml
<vfs>
  <audit>
    <fs/>
  </audit>
</vfs>
```

ehmry, 2019-04-20:
Fix at 10f03513d184fee477af64a42b3d6aaeece2af4d.

nfeske, 2019-05-06:
Even though I like the cleanliness of your suggestion, I think this change is not for the better, for the following reasons:

1. With your patch, there is no longer a natural place for `<audit>` configuration parameters. Should we want to extend the audit plugin with filtering options or a configuration of the level of detail, such configurations can no longer go into the body of the `<audit>` node. Well, they could - but this would be not _intuitive_.

2. It is no longer possible to audit a specific subdirectory of a file system but only an entire <fs>. E.g., it won't be possible to trace only the `/depot/local` directory of a rump fs.

3. Related to 2, it won't be possible to restrict the auditing to a particular client. With the original version, it is possible to use the `<policy root="...">` attribute to direct one client to the <audit> directory while directing another client to the raw directory directly.

4. The semantics of the body of the `<audit>` node are not clear cut. E.g., does it accept a single element or multiple elements (like a `<dir>` node)? How does the plugin behave in union-mount situations, e.g., when reading directory entries while the <audit> shares a path with other parts of the vfs? The original version leaves no room for such uncertainties.

@ehmry Do you deem it worth limiting the flexibility of the plugin in these respects on the account the (subjectively) improved intuitiveness?

ehmry, 2019-05-06:
Yes, in my opinion the current version is awkward enough that its flexibility is imperceptible.

This new version is just a wrapper over the the `Dir_file_system`, so it should be able to do everything the current version supports.

ehmry, 2019-05-06:
I'm fine with leaving the plugin as it is and maintaining an alternative.

closed 2019-05-06 by ehmry