genodelabs/genode #3303: window manager: jumping windows

alex-ab, 2019-04-25:
When using a nitpicker client (top_view in my case) it happens reliable that the whole client window "jumps" by several 100 pixels to the left when just clicking actually a button inside the window client. This does not happen when using it with nitpicker as server.

alex-ab, 2019-04-25:
In the attached trace one can see, that the window position (xpos) jumps from 150 to 92 in the event of the button click.

```
[init -> wm -> report_rom] report 'wm -> pointer'
[init -> wm -> report_rom]   <pointer xpos="171" ypos="247"/>                   <-- A -->
[init -> wm -> report_rom] report 'decorator -> hover'
[init -> wm -> report_rom]   <hover>
[init -> wm -> report_rom]   	<window id="1">
[init -> wm -> report_rom]   		<left_sizer/>
[init -> wm -> report_rom]   	</window>
[init -> wm -> report_rom]   </hover>
[init -> wm -> report_rom] report 'layouter -> window_layout'
[init -> wm -> report_rom]   <window_layout>
[init -> wm -> report_rom]   	<window id="1" title="menu_view -> " xpos="150" ypos="130" width="522" height="236" focused="yes"/>
[init -> wm -> report_rom]   </window_layout>
[init -> wm -> report_rom] report 'wm -> pointer'
[init -> wm -> report_rom]   <pointer xpos="174" ypos="247"/>                   <-- B -->
[init -> wm -> report_rom] report 'layouter -> window_layout'
[init -> wm -> report_rom]   <window_layout>
[init -> wm -> report_rom]   	<window id="1" title="menu_view -> " xpos="92" ypos="130" width="522" height="236" focused="yes">
[init -> wm -> report_rom]   		<highlight>
[init -> wm -> report_rom]   			<left pressed="yes"/>
[init -> wm -> report_rom]   		</highlight>
[init -> wm -> report_rom]   	</window>
[init -> wm -> report_rom]   </window_layout>
[init -> wm -> report_rom] report 'wm -> pointer'
[init -> wm -> report_rom]   <pointer xpos="174" ypos="247"/>
[init -> wm -> report_rom] report 'decorator -> hover'
[init -> wm -> report_rom]   <hover>
[init -> wm -> report_rom]   	<window id="1"/>
[init -> wm -> report_rom]   </hover>
```
The actual reason seems to be here, that the decorator did not yet processed the new mouse position (<-- B -->) and  therefore the layouter takes the old outdated report which has left_sizer set. The layouter wrongly assumes that due to the click a dragging starts which is not the case.

alex-ab, 2019-04-25:
The "wm: avoid jumping windows" is more a quirk than a fix, which works for me now much better. Due to the asynchrnous nature of the  wm<->decorator<->layouter   triple it may happen ever that the layouter sees outdated reports of the decorator, depending on scheduling and load, especially seen on Sculpt.

chelmuth, 2019-05-06:
Although this issue is marked fixed, I'm uncertain about a merge (mostly because it's qualified as quirk ;-). @nfeske what do you think?

nfeske, 2019-05-07:
Thanks @alex-ab for the investigation of this mystery. It haunted me since a long time. So your discovery is much appreciated.

@chelmuth I haven't had the time to look at it yet. Before merging the patch, I'd like to take some time to dive into the topic again.

nfeske, 2019-05-15:
Commit https://github.com/alex-ab/genode/commit/7e21febb164244d6c8f42703afef320d34a84514 looks good to me. The missing reset of the drag state is definitely an oversight of mine. Good catch, @alex-ab!

I'm afraid that commit https://github.com/alex-ab/genode/commit/c9bce28641fbb53a4d8a711006b479223a512329 covers up a symptom of a problem somewhere else. A stale hover report alone should - in my understanding - not be able to make a window jump. I'll investigate...

@alex-ab, the "outdated reports" topic is actually considered in the design. It is inherent whenever components are loosely coupled. E.g., when the window-layout report produced by the layouter arrives at the decorator late, the windows simply remain on their old positions until the decorator has picked it up. No problem. Similarly, the hover report may temporarily become out of sync with the decoration, but this should not have any different effect than a pointer movement (briefly loosing the hovering of a window element wouldn't be a problem). Not trying to downplay the issue, just an attempt to defend the design. ;-)

nfeske, 2019-05-15:
I can reproduce the jumping windows, but only with the themed_decorator. The motif decorator won't let the windows jump.

alex-ab, 2019-05-16:
Right, I'm using the themed_wm package on Sculpt.

nfeske, 2019-06-17:
The commits above fix the jumping-window problem. Please refer to the commit message of the first commit for the explanation. The second commit handles the residual corner case mentioned in the commit message of the first.

alex-ab, 2019-06-17:
Thanks for taking care :+1:

closed 2019-07-09 by chelmuth