genodelabs/genode #3304: nic_router_flood: not working for many months

skalk, 2019-04-25:
While working on issue #3180 I updated the run-script `nic_router_flood`. But I cannot really test whether it works, because the last time it run successfully on at least one platform (`nova_x86_64` only) was 1th October last year.

@m-stein To me it seems to be useless to keep the test when its result has no effect. What do you think? Does it trigger issues worth to investigate, or is the mimic of the test already targeted by one of the other nic_router centered tests?

m-stein, 2019-04-25:
@skalk 
I see your point, I've treated this test shabbily in the past. However, I think that it is a very important test as it exposes the router to clients that misbehave on network-protocol level. This makes it unique amongst the current NIC router tests. It triggers, for instance, various rare exceptions during the application of client quota. I'll have a look at it these days. But it could be that the first results don't show up before the 6th of May as I'm out of office next week.

skalk, 2019-04-26:
@m-stein fine!

m-stein, 2019-05-07:
It's broken since ae559549198449d13cbb852a72c2e2f165f6ec6d .

m-stein, 2019-05-08:
@skalk @chelmuth The issue is more complicated than I thought. Thus, I would suggest to revert ae55954, so, I'm not on a critical path with re-enabling the advanced test:
81c9b9c Revert "nic_router_flood: reworked to stress/analyze more"

Until now, I can at least tell, that these unmerged commits of mine fix some issues with the advanced test:
fad21bd heap: fix exception handling in _allocate_dataspace
44fe66d allocator_avl: fix dangling-allocations warning

chelmuth, 2019-05-08:
I merged fad21bd7abc1554ce643d0547b361523e2d0b28a just now but cannot grasp how the fix is related to this issue.

nfeske, 2019-05-08:
Commit https://github.com/genodelabs/genode/commit/fad21bd7abc1554ce643d0547b361523e2d0b28a looks good. I remember that I was hesitant to merge it because of the nested exception handling and the duplicated code in the exception handlers. But this style consideration can be improved later. The fix is good.

Commit https://github.com/genodelabs/genode/commit/44fe66d7fe59080358cc71be0f38469dcb5427d6 should not be merged in my opinion. It solves a cosmetic problem only. So it shouldn't affect the tests. Of course, I'd like to see the false-positive warnings disappear and appreciate that @m-stein created a solution. But extending the `Allocator_avl` and `Slab` interfaces goes too far. A less invasive solution would be to count the metadata allocations locally within the heap (by observing the nested calls of alloc and free) and subtract this count from the `dangling_allocations` count.

skalk, 2020-03-03:
The issue(s) regarding `nic_router_flood.run` does not occur anymore in nightly tests.

closed 2020-03-03 by skalk