genodelabs/genode #3307: tool chain update for 19.05 release

cproc, 2019-04-25:
Planned features:
- gcc version 8.3.0
- binutils version 2.32
- gdb version 8.2.1
- aarch64 support (issue #3260)
- risc-v improvements (issue #3273)
- thread/tls support (including thread-safe exceptions)
- ali2dep instead of gnatmake
- integration of the tool chain into the Genode SDK
- build Genode components with C++17 by default
- build Genode components with -O3 by default

cproc, 2019-04-25:
I'm planning to push snapshots of current work in progress to my issue3307 branch [1] in case someone needs to build the latest tool chain for testing. At the current state, the tool chain with the updated tool versions can be built with the 'tool_chain' script and I created workarounds for some errors that came up when building Genode components with this tool chain. I'm planning to create individual GitHub issues for these problems later. Currently, I'm debugging stability problems with VirtualBox which appeared when built with the updated tools.

[1] https://github.com/cproc/genode/tree/issue3307

cproc, 2019-05-07:
depends on #3322

cproc, 2019-05-07:
depends on #3324

cproc, 2019-05-07:
depends on #3325

cproc, 2019-05-07:
depends on #3326

cproc, 2019-05-07:
depends on #3327

cproc, 2019-05-07:
depends on #3328

cproc, 2019-05-07:
depends on #3330

cproc, 2019-05-07:
depends on #3331

cproc, 2019-05-07:
depends on #3332

cproc, 2019-05-07:
depends on #3333

cproc, 2019-05-07:
depends on #3334

cproc, 2019-05-07:
depends on #3335

cproc, 2019-05-09:
depends on #3341

cproc, 2019-05-09:
depends on #3342

cproc, 2019-05-09:
depends on #3343

cproc, 2019-05-09:
depends on #3344

cproc, 2019-05-09:
depends on #3345

cproc, 2019-05-09:
depends on #3347

cproc, 2019-05-09:
depends on #3348

cproc, 2019-05-14:
depends on #3350

cproc, 2019-05-14:
depends on #3351

cproc, 2019-05-14:
depends on #3352

cproc, 2019-05-20:
@chelmuth: I pushed the commits to my 'merge_to_staging' branch

ehmry, 2019-05-21:
The toolchain should probably be placed in a versioned path for this release rather than just copied over the previous toolchain, otherwise the SDK cannot enforce which toolchain to use, and it makes backporting unnecessarily difficult.

nfeske, 2019-05-22:
> The toolchain should probably be placed in a versioned path for this release rather than just copied over the previous toolchain

I agree. Let us install the tool chain at

```
/usr/local/genode/tool/19.05/ { bin lib share ...}
```

In addition, we should create a symbolic link at `/usr/local/genode/tool/current` that points to `19.05`. This makes it easy to let tools like ccache pick up the "current" tool chain.

This change will require the adaptation of the `CROSS_DEV_PREFIX` in `repos/base/etc/tools.conf` (using the versioned path). This is good because the tool chain is then implicitly switched when changing branches, unless `CROSS_DEV_PREFIX` is defined manually.

This this directory layout in place, we can install the sdk later into `/usr/local/genode/sdk/<version>`.

ehmry, 2019-05-22:
I have a patch at a3ef9285f9814caa2d85a802ae1438b4cd591a84 that allows OpenTTD to link again (OpenTTD doesn't work, but I build it anyway).

cproc, 2019-05-24:
Commit eb14550 implements the versioned install location.

cproc, 2019-05-24:
Commit 3f2772f implements the versioned install location without changing the outdated sel4 documentation.

nfeske, 2019-07-09:
Part of Genode 19.05

closed 2019-07-09 by nfeske