genodelabs/genode #3308: spunky: a kernel using Ada

m-stein, 2019-04-26:
I have started this in my spare time as a low-prio project. The goal is to see how far I can get replacing C++ code from the base-hw kernel with Ada code. I'll successively pick single aspects of the kernel and translate them in-place by using the approach for C++/Ada interfacing presented in https://github.com/genodelabs/genode/commit/98182379188fab9f262a8df8623d9fa4843a0ddc. Eventually this could become a new kernel besides base-hw but I'm not sure if this is desired. I'm not aiming for SPARK in the first phase.

m-stein, 2019-04-26:
There is a version that uses Ada for the RPC state machine and succeeds in the Init test on 32/64 bit:
https://github.com/m-stein/genode/commits/3308_spunky_a_kernel_using_ada

m-stein, 2019-04-30:
@senier The second time now, I encountered this compilation error that I cannot interpret:
```
.../contrib/ada-runtime-0e4625502d44eda1974fa447601aadd6b843e067/ada-runtime/contrib/gcc-6.3.0/s-finmas.ads:32:06: run-time library configuration error
.../contrib/ada-runtime-0e4625502d44eda1974fa447601aadd6b843e067/ada-runtime/contrib/gcc-6.3.0/s-finmas.ads:32:06: file s-finmas.ads had semantic errors
.../contrib/ada-runtime-0e4625502d44eda1974fa447601aadd6b843e067/ada-runtime/contrib/gcc-6.3.0/s-finmas.ads:32:06: entity "System.Finalization_Masters.Finalization_Master" not available
.../contrib/ada-runtime-0e4625502d44eda1974fa447601aadd6b843e067/ada-runtime/contrib/gcc-6.3.0/s-finroo.ads:40:09: run-time configuration error
.../contrib/ada-runtime-0e4625502d44eda1974fa447601aadd6b843e067/ada-runtime/contrib/gcc-6.3.0/s-finroo.ads:40:09: entity "System.Soft_Links.Abort_Undefer" not defined
```
The change I have made to trigger the error was of this kind:
#### Good version ####
```
pragma Ada 2012;
package Master is
   ...
private
   package Sub is ... end Sub;
end Master;
package body Master is
   ...
   package body Sub is ... end Sub;
end Master;
```
#### Bad version ####
```
pragma Ada 2012;
package Master is
   ...
private
   package Sub ... end Sub;
   package Copy_Of_Sub is ... end Copy_Of_Sub;
end Master;
package body Master is
   ...
   package body Sub is ... end Sub;
   package body Copy_Of_Sub is ... end Copy_Of_Sub;
end Master;
```
The first time, Sub was a generic that I tried to instantiate for a specific type. I solved the error by not using generics but a specific package.
Is this a known problem to you?

m-stein, 2019-04-30:
@senier The problem seems to be related to a real error in my code. My Master package contained type declarations
```
type A is private;
type A_Ref is not null access all A;
```
And ``A_Ref`` was used by ``Copy_Of_Sub`` but I forgot to define ``A`` in the private part. Instead of pointing me to this, the compiler showed the above mentioned errors. As soon as I added the definition of ``A``, it worked.

m-stein, 2019-04-30:
I tried the generic queue example again as I'd like to get rid of my redundant queue implementations but the errors about the runtime configuration persist. This minimal example triggers them:
```
pragma Ada_2012;
package Shop is

   type Customer_Type            is private;
   type Customer_Reference_Type  is not null access all Customer_Type;

private

   generic
      type Payload_Type;
      type Payload_Reference_Type is not null access all Payload_Type;
   package Generic_Queue is

      type Queue_Type        is private;
      type Item_Type         is private;
      type Item_Pointer_Type is access all Item_Type;

   private

      type Queue_Type is record
         Head : Item_Pointer_Type;
         Tail : Item_Pointer_Type;
      end record;

      type Item_Type is record
         Next    : Item_Pointer_Type;
         Payload : Payload_Reference_Type;
      end record;

   end Generic_Queue;

   package Customer_Queue is new Generic_Queue (
      Customer_Type,
      Customer_Reference_Type);

   type Customer_Type is record
      Queue_Item : aliased Customer_Queue.Item_Type;
      Order      : Integer;
   end record;

end Shop;
```
As soon as I replace the generic queue with a similar specialized version, it compiles.

senier, 2019-04-30:
The very same error as above?

Can you try a couple of things (separately) an report whether that makes any difference:

1. Remove the `not null`
2. Make `Customer_Type` public

m-stein, 2019-04-30:
This is the error with the generic queue:
```
.../contrib/ada-runtime-0e4625502d44eda1974fa447601aadd6b843e067/ada-runtime/contrib/gcc-6.3.0/s-finmas.ads:32:06: run-time library configuration error
.../contrib/ada-runtime-0e4625502d44eda1974fa447601aadd6b843e067/ada-runtime/contrib/gcc-6.3.0/s-finmas.ads:32:06: file s-finmas.ads had semantic errors
.../contrib/ada-runtime-0e4625502d44eda1974fa447601aadd6b843e067/ada-runtime/contrib/gcc-6.3.0/s-finmas.ads:32:06: entity "System.Finalization_Masters.Finalization_Master" not available
.../contrib/ada-runtime-0e4625502d44eda1974fa447601aadd6b843e067/ada-runtime/contrib/gcc-6.3.0/s-finroo.ads:40:09: run-time configuration error
.../contrib/ada-runtime-0e4625502d44eda1974fa447601aadd6b843e067/ada-runtime/contrib/gcc-6.3.0/s-finroo.ads:40:09: entity "System.Soft_Links.Abort_Undefer" not defined
```
I tried your suggestions, they led me to a simpler design:
```
pragma Ada_2012;
generic
   type Payload_Type;
   type Payload_Reference_Type is access all Payload_Type;
package Generic_Queue is
   type Queue_Type        is private;
   type Item_Type         is private;
   type Item_Pointer_Type is access all Item_Type;
private
   type Queue_Type is record
      Head : Item_Pointer_Type;
      Tail : Item_Pointer_Type;
   end record;
   type Item_Type is record
      Next    : Item_Pointer_Type;
      Payload : Payload_Reference_Type;
   end record;
end Generic_Queue;
---------------------------------------------
pragma Ada_2012;
with Generic_Queue;
package Shop is
   type Customer_Type is record
--    Queue_Item : aliased Customer_Queue.Item_Type;  --  would be bad
      Order      : Integer;
   end record;
   type Customer_Reference_Type is access all Customer_Type;
   package Customer_Queue is new Generic_Queue (
      Customer_Type,
      Customer_Reference_Type);
end Shop;
```
I could only make it work only by moving the customer definition above the queue specialization. A forward declaration produces the above mentioned error. But without, I can't have the ``Queue_Item`` field in ``Customer_Type``. Maybe this isn't possible with Ada 2012 after all.

m-stein, 2019-05-01:
@senier Btw. both of your suggestions independent from each other did'nt make a difference with the original example.

senier, 2019-05-21:
Unfortunately, defining a generic access type pulls in `System.Finalization_Masters` which is very complex as it is a controlled type (which is based on tagged types). Implementing support for finalization would require reimplementing `Finalization_Masters` without using controlled or tagged types. We don't believe this is feasible (or worthwhile) for our runtime. Sorry, for the time being this construct is unsupported.

m-stein, 2019-05-21:
@senier Thanks for your profound explanation! So, am I right that the problem is that support for tagged and controlled types is missing?

jklmnn, 2019-05-21:
Basically. There are two options:
 - implementing support for tagged and controlled types, which includes full dynamic dispatching for tagged types
 - reimplementing `System.Finalization_Masters` without using controlled types

Both options require a thorough understanding of the mechanisms behind this functionality and the added complexity will likely exceed the runtimes current complexity.

m-stein, 2019-05-21:
Ah ok, so, I understood it right :) 
From the comment of @senier that you believe that the second option is not feasible or very complicated, I conclude that the first option would be the more preferable. Support for controlled and tagged types would be desirable in the long term anyway.

senier, 2019-05-21:
In fact, we believe that both options are infeasible. We tried copying tagged type support from the full runtime earlier. This resulted in manifold increase of the code base.

Reimplementing it without controlled types may not be viable at all. I guess they used it for reason and understanding that part of the runtime will be very challenging.

jklmnn, 2019-05-27:
You could try an ADC file with [`pragma Restrictions (No_Finalization)`](https://docs.adacore.com/gnat_rm-docs/html/gnat_rm/gnat_rm/standard_and_implementation_defined_restrictions.html#no-finalization) that seems to disable the mechanisms that cause the problems here. Although it might happen that your code gets rejected by the compiler then, too.

m-stein, 2019-05-28:
@jklmnn @alex-ab Ah, thanks for sticking to this issue so persistent! I'll try the pragma as soon as I have some spare time and provide feedback then!

m-stein, 2019-05-28:
I meant @senier and not alex in the last commit ^^

m-stein, 2019-05-28:
@jklmnn @senier Awesome :-) the pragma solved the issue completely. I can now use generic queues which saves a lot of redundancy. Thanks again for investigating!

senier, 2019-05-29:
Great to hear this worked for you!

m-stein, 2019-12-20:
I've pushed a re-based version that has signaling and RPC implemented in Ada and moves Spunky to its own base-repository. There are several minor adaptions to base-hw motivated by Spunky that I would suggest for staging. Besides simplifying things for Spunky, they are partly also beneficial for base-hw.

@skalk Would you mind doing a review fof the base-hw commits?

44af66e063 base-spunky: tooling + RPC and signals in Ada
80e777b62e base-hw: hook to prepare first execution of core
c24906f5f6 base-hw: no method lambdas in synchronous IPC
18d4994ea0 base-hw: do not use Signal::Data in signaling
022f6b1fcb base-hw: no error return codes in signaling
65502990bf base-hw: obsolete Kernel::Thread::_signal_receiver
fdbd5a2cdc base-hw: no pointer arguments in signaling
12959416bd base-hw: do not inherite Kernel::Object
94b8b43cfc base-hw: no virtual functions in signaling
d66a595608 base-hw: Ada/SPARK-friendly Ipc_node interface
cd51f26ae9 base-hw: remove unused Ipc_node accessor methods
da9026eaa3 base-hw: avoid that Ipc_node inherits
9609fead68 base-hw: no PD code / virt methods in Ipc_node

m-stein, 2020-02-06:
@chelmuth @skalk: I've cleaned up the base-hw preparation series according to Stefans review (thanks at this point!) and pushed the commits that I think are sensible for staging (all except 29e3505092, 50d8828fc2, 5ab059b08f) with my ``merge_to_staging``:
7b086ef0d2 base-hw: no pointers in scheduler interface
a74841c5f5 base-hw: cpu scheduler without inheritance
e65e5fe834 base-hw: double list without inheritance
f879a96ccc base-hw: fix run scripts
84eeeec3ab base-hw: no error return codes in signaling
3df12a4356 base-hw: obsolete Kernel::Thread::_signal_receiver
6cec7d63a4 base-hw: no pointer arguments in signaling
6a8e5f264b base-hw: do not inherit Kernel::Object
ce1e91eded base-hw: no virtual functions in signaling
c90858edbb base-hw: Ada/SPARK-friendly Ipc_node interface
9f3d054585 base-hw: remove unused Ipc_node accessor methods
f4729cb996 base-hw: avoid that Ipc_node inherits
7fc49a5051 base-hw: no PD code / virt methods in Ipc_node

m-stein, 2020-02-07:
@chelmuth @skalk I have changed and rebased the commit series once more:

edf68ac6d1 base-hw: Ada-compliant scheduler method signatures
c7e599db34 base-hw: no pointers in scheduler interface
2858051ee3 base-hw: cpu scheduler without inheritance
a59b20bc5c base-hw: double list without inheritance
7f3908e4be base-hw: fix run scripts
db3143fa82 base-hw: no error return codes in signaling
fa7ed27120 base-hw: obsolete Kernel::Thread::_signal_receiver
6b9bd30e94 base-hw: no pointer arguments in signaling
8c02c66253 base-hw: do not inherit Kernel::Object
98608a81c2 base-hw: no virtual functions in signaling
b9dfb37047 base-hw: Ada/SPARK-friendly Ipc_node interface
e653813a87 base-hw: remove unused Ipc_node accessor methods
507c5d69a6 base-hw: avoid that Ipc_node inherits
f4045ce408 base-hw: no PD code / virt methods in Ipc_node

skalk, 2020-02-07:
@m-stein @chelmuth I'm fine with those changes, or better I like them a lot!

m-stein, 2021-05-31:
After a very long time in stand-by, I made progress regarding this issue again. Spunky has now a timer driver / timeout scheduler completely written in Ada. I think the next step will be the interrupt controller.

m-stein, 2021-06-07:
I'm currently debugging the freshly translated IRQ controller. The findings of the day:
* the kernel gets initialized correctly and the userland starts, but after the first timer interrupt, when trying to return to the userland, qemu gets stuck for arbitrary reasons like access outside of memory or a spontaneous reboot
* I checked that I receive correct function parameters in Spunky, so, the interfacing between C++ and Ada doesn't seem to be the problem. 
* I checked that the MMIO registers have the same virtual addresses as in base-hw
* I made those member variables that are static in base-hw also "static" in Spunky, didn't fix it
* I checked that the controller initialization writes the same register values as in base-hw
* Changing the number of CPUs to 1 doesn't alter the issue, so, it's not a problem of multiple controller instances

jt9897253, 2021-10-10:
Hello,

first of all, thanks to everyone involved in Genode for the work put into this project. Also, I like the idea of replacing C with safer languages.

However, by choosing Ada/Spark I wonder how such a system could be bootstrapped from source, as described on https://www.bootstrappable.org

Is there a way to build an Ada compiler without already having a binary Ada compiler?

m-stein, 2021-12-07:
Hi @jt9897253,
The concept I read on bootstrappable.org sound sensible and interesting, however, right now, this is way out of my focus. I'm not an expert regarding Ada/SPARK and especially not the tooling behind it. I'm also just learning a lot of stuff ad-hoc while developing Spunky and implementing the kernel logic alone feels enough of a challenge for me. But I would never like to discourage you from getting your hands on the bootstrappable approach yourself! I would be very interested the outcome of such an endeavor.

m-stein, 2022-09-02:
@nfeske To be honest, I have grown a bit unsure about this project. To me, our offline discussion about another Ada/SPARK project with Genode applies as well to Spunky and the question arises, which perspective has such a fundamental but "foreign-language" project in the Genode community. Furthermore, I recently prioritized other work and my private life a lot over going on with Spunky and the near future doesn't seem to make a difference.

That said, the major problem is that the project becomes more and more incompatible with Genode mainline over time and there's no one but me to keep up with the pace. This would presumably not change by merging it in the current state (it's working kernel) but only when it becomes independent from base-hw. However, this is still some way to go. Sadly, my conclusion would be to lay it to rest (closing the issue) for now but keep the branches in case that we or someone else wants to revive it later.

nfeske, 2022-09-02:
I share the sentiment you expressed in your comment. We tried pretty seriously to establish Ada as part of our tool box since the beginning of 2019 (e.g., via my article https://genodians.org/nfeske/2019-04-08-c++-spark-continuum). I found various aspects - in particular the strong typing - of the language intriguing. However, after three years have passed, I made two observations.

First, we failed to spread the enthusiasm for the language to other members of Genode's developer team. Not that we haven't tried. But most of our day-to-day challenges are unrelated to language choices. I'm thinking of challenges like bringing Chromium web engine to life, the composition of Sculpt OS on the PinePhone, the porting of large driver stacks from the Linux kernel, finding and optimizing performance hot spots throughout the system, or wrestling with quirky hardware details. Most members of our team could not see how Ada would help them with such challenges. This is because there is not tangible connection. So our evangelism was probably perceived as mentally stimulating at best, or as distracting at worst. In contrast to our joint long-year cultivated safety-oriented dialect of C++ (https://genodians.org/nfeske/2019-01-22-conscious-c++) that is ingrained as mother-tongue in everyone's mind, the use of Ada remained foreign to most of us, severely impeding the chances for team play. Our project is - above all - a team effort. When weighting the benefits of Ada against its segregating effects, I have become more and more cautious over time.

Second, it's not all roses. Yes, the language brings advantages. But on the other hand, there are many paper cuts as well. Friction caused by the tooling. Or the opaqueness of ali files that prevent us from cleanly separating ABIs from library implementations. The reliance on global constructors. The surprising complexity of the runtime even when using only SPARK. In short, the language adds benefits but also more uncertainties and maintenance costs than I anticipated.

Given these reflections, I support your decision.

m-stein, 2022-09-02:
@nfeske Thanks for posting your opinion and the related references! I agree with all what you just said and will close the issue for now.

However, if anyone is eager to continue the effort - please don't feel discouraged by our decision! You can always contact me and I'd be happy to provide you with the pointers on how to revive Spunky!

closed 2022-09-02 by m-stein