genodelabs/genode #3310: noux_terminal_fs.run does not work correctly anymore

cproc, 2019-04-28:
When running 'head -n 1 /dev/terminal' in the upper terminal, then selecting the lower terminal and entering text (finished with return key), the upper terminal does not show the entered text. The problem seems to be that the noux 'read' syscall for /dev/terminal does not get unblocked anymore when the terminal VFS plugin receives a read-avail signal from the terminal. 'noux_gdb.run' shows similar symptoms. According to git bisect the problem seems to be related to the recent VFS response handler changes (commit a635873568e2c7292f95dd3087baa09be8515a8a).

trimpim, 2019-11-25:
@chelmuth  and @nfeske I have a similar problem that my JAVA application that listens on stdin for data doesn't wake up when my other component writes data to the terminal.

I have created a depot_autopilot test that demonstrates the issue. [stdin_blockint_read](https://github.com/trimpim/genode/tree/stdin_blocking_read)

chelmuth, 2019-11-25:
@trimpim IMO the problem you describe does not fit this issue as it is not related to Noux.

Anyway, I had a deeper look into your test case and think I spotted a bug in your test server that may be the original issue. In https://github.com/trimpim/genode/blob/6b0387d1d0f6c6ae7733a9f81c7200a471fa2077/repos/libports/src/test/libc_stdin_blocking_read/server/terminal_session_component.h#L108-L117 the server fills the write buffer with data to be read by the client, but the client is not informed about this fact, i.e., the code misses `Signal_transmitter(_read_avail_sigh).submit();` to deblock blocked clients. Does this help to solve your issue?

trimpim, 2019-11-26:
@chelmuth thanks a lot, this fixes my error indeed.

chelmuth, 2019-11-27:
Sorry, the issue close was not intended :-(

nfeske, 2020-04-23:
Noux is no more.

closed 2020-04-23 by nfeske