genodelabs/genode #3312: Convert drivers from Nic servers to Nic client

ehmry, 2019-04-30:
Nic drivers are at present implemented as Nic servers. This is intuitive for static systems, but complicates the role of components such as the *nic_router* which support multiple hardware uplinks and hot swapping. If drivers were implemented as clients, then adding and removing drivers becomes for intuitive for dynamic systems centered around service multiplexers like *nic_router*.

ehmry, 2019-04-30:
I have a patch to make changes to the Nic session at 42e06cc169d7465aa6d8ac224f7315e8d7ea09ad. This adds an optional MAC address argument to the `Nic::Connection` constructor and adds getters and setters for session link-state.

I have not yet modified drivers to act as clients, only to use the new session interface.

nfeske, 2019-05-02:
This looks excellent. The second constructor is an elegant solution. So the server knows right at session creation time the role of the client. Very nice! I also appreciate the thoughtful commit message, giving good context information.

ehmry, 2019-05-06:
@m-stein if you would like to continue this work please use my `nic_session` branch as a base, which should only the 2d11821aa9d82bc32a56deba61e4fa5dd7e12af9 and c58b4a478f191ca6e1aca5cc370d93891dad016d patches.

m-stein, 2019-05-06:
@ehmry Thanks

ehmry, 2019-05-14:
More patches for this issue, 83e6c2e8b93f148fb5e8dcb506e586e4d2070266, 92abbc88a96fcea0d80747b3e8a22df09c13e3c0, 3d3e7e182a4f227ef645603a3968d20e177cddcf, and 6982004a27c1e0c7c5cf84329946128d28255cd1.

The USB nic driver still needs conversion.

nfeske, 2019-05-15:
@ehmry That's great progress. But I'd like to shift gears down for a bit. Even though we planned the "pluggable NIC drivers" for the 19.05 release, I would prefer to defer the topic to the next release cycle. This would give us more time to plan the transition to the benefit of Genode's users.

With your current patch series, the change is disruptive. I'd rather like to introduce the change incrementally. In particular, the client-side role of the drivers should first become an option. Granted, in this step, the drivers become inflated a fair bit. But this is just temporary. Then, we can adjust the scenarios step by step. No need to make one swift change covering everything at once. This approach alleviates stress on our side. Once all scenarios are comfortably covered, we can drop the server-side role from the drivers.

Regarding the change of the NIC multiplexers, we may consider scenarios where those components may be used in a cascaded fashion. Here, the client role for the uplink may still be handy.

As a small remark regarding naming, we have used the term "uplink" until now. I'd like to keep using this name consistently instead of introducing a second one ("upstream").

m-stein, 2019-05-15:
I agree with what @nfeske just wrote.

ehmry, 2019-05-20:
Would it be acceptable to maintain separate server and client driver implementations during the transition? Selecting the proper driver would effectively be a configuration task, and this would avoid implementing the configuration logic at first and untangling the drivers later.

nfeske, 2019-05-21:
I would not duplicate the components to avoid the need to introduce a scheme for naming components. @ehmry you are right that we should avoid entangling the drivers with much additional logic, but I think the following - or a similarly implemented - distinction directly in the `Component::construct` function would be acceptable (just a sketch):

```
if (Genode::Attached_rom_dataspace(env, "config").xml().attribute_value("server", true)) {
  static Server_main main(env);
} else {
  static Main main(env); /* client role */
}
```

Here `Main` would implement the client-role implementation whereas `Server_main` would implement the server-role implementation. Once the transition is complete, we can remove the `Server_main` class and the conditional construction in `Component::construct`.

ehmry, 2019-05-29:
I think combining server-style and client-style drivers into single components as proposed by @nfeske is trivial to implement at this point. However, the post-release situation is quite different from when this issue was opened because bare, platform-specific drivers have been replaced in runscripts by abstract driver packages (#2190). This begs the question of what should be the role of the driver packages, and if the drivers should support dual-role, or incrementally combined with the nic_bridge or nic_router servers within the driver packages, and if this is the case, what are the implications for runscripts that test the nic_router or nic_bridge directly.

nfeske, 2019-06-03:
Thanks @ehmry for initiating the discussion.

I think the drivers_nic package should have as little policy as possible but enough policy to be useful. The line could be drawn similar to the drivers-interactive pkgs where nitpicker is not part of the pkg. 

Following this line of thinking, I would keep the NIC router out of the drivers_nic packages.
The drivers package should not support both client and server roles. Currently, the pkgs provide a Nic service. But at one point in the not-too-distant future, we should make a switch towards the client-side role. We could try doing this switch incrementally as follows.

1. First, we would introduce a NIC router into all scenarios that don't already have one.
2. We add a "Nic" route from the drivers_nic subsystem to the NIC router. But this route remains unused as long as the drivers pkgs have the server role. But the added route won't break anything.
3. We add a policy to the NIC router that assigns Nic clients originating from the drivers_nic subsystem to the "uplink". Still, right here, this policy is never used. But like (2) it does not break anything.

With these preparations in place, the run scripts should in principle be compatible with both a client-side and server-side role of the drivers subsystem.

When making the switch for a given drivers pkg, we have a situation where the NIC router still tries to open an "uplink" session at the drivers subsystem. To make the NIC router happy, the drivers pkg could (temporarily - while we are in the transition) host a dummy NIC service. But the real traffic will now go through the client-side role of the drivers pkg.

Once all drivers_nic pkgs are adjusted, we can enter the cleanup phase.

1. Remove the old "uplink" nodes from the NIC-router configurations.
2. Remove the dummy NIC service from the drivers_nic pkgs.
3. Remove the "Nic" route from the NIC router to the drivers_nic subsystem.

Does this sound like a feasible plan?

m-stein, 2019-06-03:
@nfeske Thanks for working out such a detailed plan. To me it sounds sensible. I'd like to add only a little suggestion: How about not adding an explicit route to NIC router to the drivers package but instead a route to a more generic name like "drivers_nic_server" and then add aliases like ``<alias name="drivers_nic_server" child="nic_router">`` to the tests? This would give us more flexibility regarding the actual NIC server.

nfeske, 2019-06-03:
> How about not adding an explicit route to NIC router to the drivers package but instead a route to a more generic name like "drivers_nic_server"

@m-stein I'm not sure I understand your suggestion completely.

Inside the drivers pkg, there wouldn't be any reference to the NIC router. Only in the actual run script, there would be a route from the "drivers" start node to the "nic_router" start node. So the route remains local to each run script. I think the alias you suggested won't make things any simpler. I think it would be perfectly fine to speak of the "nic_router" in run scripts to refer to the NIC server, similar to how we refer to nitpicker in as the GUI server in many interactive run scripts.

m-stein, 2019-06-03:
> Inside the drivers pkg, there wouldn't be any reference to the NIC router. Only in the actual run script, there would be a route from the "drivers" start node to the "nic_router" start node.

Ah, ok, that was the point I missed. So, my suggestion doesn't make sense.

m-stein, 2021-01-19:
This issue was replaced by #3961 . So, I'd vote for closing it.

closed 2021-01-19 by m-stein