genodelabs/genode #3313: sculpt: use block_cache to improve performance

NobodyIII, 2019-05-02:
Currently, Sculpt has terrible I/O performance when running from a flash drive. I suspect that using block_cache would improve performance dramatically for that use case, and would also provide a reasonable performance boost for HDDs and even SSDs.

The main question in my mind isn't whether to do this, but how to best determine the RAM quota(s). I suspect that 128 MiB would provide a huge performance increase without using up too much of the RAM.

Anyone else have any thoughts on this?

ehmry, 2019-05-02:
The performance bottleneck is probably in the rump file-system plugin. Rump has its own block cache, so I don't think anyone has tried to combine rump with block_cache.

NobodyIII, 2019-05-02:
That makes sense. The very slow speed made me think rump_fs wasn't caching
much, if anything. Then I think I'll try increasing the cache size. That
might help. On normal OSes, it's common for disk caches to grow to
gigabytes if there's a lot of RAM. I'll try 128M and see how much of a
difference that makes.

nfeske, 2019-05-03:
@NobodyIII i agree with everything you said. Your observation is the reason we created the block-cache component in the first place. However, it remained unused in practice until today. So unlike the components currently used by Sculpt it has not stood the test of time. Hence, I feel quite uncertain about it. I think that adding a component won't cut it. In order to add good block caching into Sculpt, we need to look at the problem more holistically.

* The sculpt manager must support the ballooning of selected components (like the block cache). This includes the balancing of resources between multiple balloons. Think of multiple block devices, each with a block cache.
* The block cache should definitely be reworked to use the new block-level API #3283 to benefit from the performance boost of the batching of requests and the robustness improvements given by the new API.
* We must weight the overhead of putting the block cache into a dedicated address space against the gained security benefit over co-locating it with the file system. We may potentially come to the conclusion that the current multi-component approach yields no security benefit but adds performance costs. Hence, it might be worthwhile to contemplate the idea of implementing the block cache as a VFS plugin instead, and change the file-system VFS plugins to access the block device via the VFS instead of a block session. This would be a major topic.

We should definitely explore these areas. But given their reach, they are not for today. At present, I'd instead suggest to look at the current storage stack and try to optimize it in other areas first. I'm convinced that the performance bottleneck you observed is not only on account of insufficient caching but there are other deficiencies as well. We should concentrate on finding and fixing them. This involves - as a first and fundamental step - the gathering of data. Until now, we have not pursued systematic measurements of the storage stack (except for the lowest layer). So this should be our priority.

The bottom line is that we won't take steps to address this issue in the near future. To avoid the issue from lingering around, I suggest to close it and encourage you to experiment with the block cache independently.

NobodyIII, 2019-05-03:
Okay. Sounds reasonable. It seems like benchmarking different parts of the
storage stack could be very useful. I'll look into that. And streamlined
RAM ballooning would be very nice. I'll try to pick up where I left off on
turning init into a library, to make ballooning easier and faster.

nfeske, 2020-04-23:
I'm closing this issue because it is rather superficial and has become stale.

Even though I agree that Sculpt's I/O performance could be optimized, we should address this topic in multiple - tangible - steps, opening individual issues once we plan to pursue actual work.

closed 2020-04-23 by nfeske