genodelabs/genode #3315: Pthread scheduling ambiguity

ehmry, 2019-05-03:
I've been working on porting [OpenTTD](https://www.openttd.org/) in the evening and I've encountering a deadlock in the main loop. OpenTTD seems to be a worthwhile port because it has a codebase that has stabilized over a 15 year period.

It seems that OpenTTD creates a pthread and then immediately blocks on a semaphore for a signal from the pthread. With Genode 19.02 it seems that the pthread is created and preempts the main thread, signalling the semaphore before the main thread begins to block, and then both threads sit blocked on the semaphore.

https://github.com/OpenTTD/OpenTTD/blob/3b8db31a49b0f1b86e3bb8d8ef4233868fcd4951/src/video/sdl_v.cpp#L180
https://github.com/OpenTTD/OpenTTD/blob/3b8db31a49b0f1b86e3bb8d8ef4233868fcd4951/src/video/sdl_v.cpp#L689

Perhaps when pthreads are created they should immediately suspend once to behave as expected?

ehmry, 2019-05-03:
So much for stability, when I checked out the repository earlier this week it was pthreads, now it's std::thread.

cnuke, 2019-05-03:
Suspending a pthread on creation is not something you normally do (if that is at all possible on all platforms). Looking into OpenTTDs Thread* implemenation it looks like they use `pthread_cond_*` underneath - chances are you stumple upon a problem with our implementation.

chelmuth, 2019-05-03:
Maybe I'm wrong but how could the main thread (or better both threads) block on a semaphore the pthread signalled already?

chelmuth, 2019-11-05:
Can we close this issue? Or, could you test with current staging again?

closed 2019-11-05 by ehmry