genodelabs/genode #3316: Deprecate board-specific build directories

skalk, 2019-05-03:
Now, that we have issue #2190 solved, we can use architecture-specific build directories only, to build all components. This stays in line with the depot packages which are only dependent on the hardware architecture too. The only issue left is that the majority of run-scripts is dependent of the board's SPEC. To solve this, we could use a BOARD variable analoque to the KERNEL variable, like:
```
build/arm_v7> make run/demo KERNEL=hw BOARD=pbxa9
```

tomga, 2019-05-04:
Do I correctly understand that for this issue you will implement building distinct bootstrap and core for hw kernel in one build directory? Will the be differently named similarly to driver binaries in #2190?

skalk, 2019-05-06:
@tomga you are right that bootstrap and core libraries have to be named according to the used board when sharing one build directory. The actual core and bootstrap binaries for hw are carrying the boot modules used in a scenario and end up in the `var/run` directory of the corresponding scenario. They get overriden whenever the run-script gets invoked, so I think we do not need to name them differently in there.

chelmuth, 2019-05-09:
Beside my remark about muen I'm fine with both commits and like to enter the integration phase. @nfeske do you object? @skalk AFAICS we do not have to adapt our CI as the changes are backwards-compatible, right?

nfeske, 2019-05-09:
No objection from my side. Thanks a lot @skalk for this huge effort!

skalk, 2019-05-09:
@chelmuth @nfeske: thanks for the review, and sorry for overseeing that I commited a preliminary result. I've now updated my merge_to_staging branch accordingly. Now, everything should work, and should be downward compatible. Please, re-read especially the first of the two commits again.

commit 1b5a0b45 (2019-05-09): kernel: differentiate board-specific components

Components like kernel, core, and bootstrap that are built for a
specific board need to reside inside the same architectural dependent
build directory. For instance there are sel4, foc, and hw kernel builds
for imx6q_sabrelite and imx7d_sabre, which have to reside inside the same
arm_v7 build directory.
This commit names those components explicitely, and adapts the run-tool to it.

Fix #3316
commit a1e70b9b (2019-05-27): kernel: differentiate board-specific components

Components like kernel, core, and bootstrap that are built for a
specific board need to reside inside the same architectural dependent
build directory. For instance there are sel4, foc, and hw kernel builds
for imx6q_sabrelite and imx7d_sabre, which have to reside inside the same
arm_v7 build directory.
This commit names those components explicitely, and adapts the run-tool to it.

Fix #3316
closed 2019-05-29 by chelmuth