genodelabs/genode #3317: presence of M.2 NVMe SSD breaks platform_drv PCI reporting on AMD Ryzen system

NobodyIII, 2019-05-05:
I recently installed an NVMe 1.2 M.2 SSD in my AMD Ryzen-based desktop computer. I discovered that Sculpt doesn't detect it, and that with it installed, the platform driver fails to report any PCI devices, which also makes the WiFi option unavailable. What might be causing this, and what can I try to diagnose/fix it?

NobodyIII, 2019-05-05:
To clarify, I do get the pci_devices report, but all it has is `<pci/>`.

NobodyIII, 2019-05-06:
I examined this further, and found that along with USB working, I could get the NVMe driver to detect and mount my SSD when I force-enabled the driver. So the problem seems to be some slight difference between the PCI detection when reporting devices and when the drivers search for the devices.

NobodyIII, 2019-05-06:
It turns out that the error was actually a simple variable scope issue. The platform driver uses an Expanding_reporter, which regenerates the report if it was larger than the initial report size. The device BDF counters were initialized outside of the lambda function used to generate the report, so they weren't reset when the report was regenerated, which made the new report start after the last PCI device. This led to an empty report.

So the only real difference that adding the M.2 SSD made was adding another PCI device to report, which made the report too big. Which, ironically enough, made the final report completely empty.

NobodyIII, 2019-05-06:
By the way, I realized that I had spaces instead of tabs. For my job, I'm expected to use spaces. For Genode, I'm expected to use tabs. I created a fixup commit: 5fa6a103674babaa05adb8e1a74ecfe1f0fe8353

nfeske, 2019-05-06:
Thank you for reporting, investigating, and fixing this issue. I appreciate it a lot!

NobodyIII, 2019-05-07:
You're welcome. I'm glad to have it working now. :)

commit 87f74a40 (2019-05-06): platform_drv/x86: fix report for many PCI devices

When there are too many PCI devices, the Expanding_reporter regenerates
the report. However, this doesn't reset the BDF counter used to iterate
over the devices. This results in starting the new report after the PCI
device that triggered the report buffer overflow. This commit fixes the
issue by putting the BDF counter initialization inside the lambda
function used to generate the report.

Fixes #3317
commit 5f0aa161 (2019-05-16): platform_drv/x86: fix report for many PCI devices

When there are too many PCI devices, the Expanding_reporter regenerates
the report. However, this doesn't reset the BDF counter used to iterate
over the devices. This results in starting the new report after the PCI
device that triggered the report buffer overflow. This commit fixes the
issue by putting the BDF counter initialization inside the lambda
function used to generate the report.

Fixes #3317
closed 2019-05-07 by NobodyIII