genodelabs/genode #3323: Xml_node::value(T &out) is missing

ValiValpas, 2019-05-07:
I think commit 5f1f67153b40a11d45f9e80de9e9ad70dc3e333e accidentally removed the Xml_node::value(T &out) method. Note that the unsafe version Xml_node::value(T *out) is still present.

nfeske, 2019-05-07:
It's actually not quite an accident. ;-)

I preserved the original `value(T *out)` method merely to keep compatibility for now, so we can transition away from the `value` method without breaking things immediately.

The `Xml_node::value` method is meant to be removed. The reasoning is that values should generally be stored in attributes, not nodes. Normally, if a node contains anything, it should be other nodes. In the rare case where a node contains something else, the `with_raw_content` method ought to be used to access it.

ValiValpas, 2019-05-07:
Alright, I was just wondering why the implementation of `value(T *out)` forwards the call to the non-existent `value(T &out)`.

Unfortunately, there are XML schemas around in which nodes contain actual data, but I can get along with using `with_raw_content`.

nfeske, 2019-05-07:
> Alright, I was just wondering why the implementation of value(T *out) forwards the call to the non-existent value(T &out).

Oops. _That_ is an accident for sure. So long for the idea of a smooth API transition...

commit 64ee151d (2019-05-23): Refinements for updated "Genode Foundations" book

- Improve API descriptions
- Remove obsolete Xml_node::value method (fixes #3323)
- Follow coding style 'const char' -> 'char const'
closed 2019-05-29 by chelmuth