genodelabs/genode #3326: base-hw: compile errors with GCC 8.3.0

cproc, 2019-05-07:
When building 'make KERNEL=hw LIB=bootstrap-hw' with GCC 8.3.0 for x86_64, the following error occurs:

```
    COMPILE  bootstrap/spec/x86_64/platform.o
/.../repos/base/../base-hw/src/bootstrap/spec/x86_64/platform.cc: In constructor ‘Bootstrap::Platform::Board::Board()’:
/.../repos/base/../base-hw/src/bootstrap/spec/x86_64/platform.cc:60:9: error: ‘invalid’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  return invalid;
         ^~~~~~~
```

cproc, 2019-05-09:
The commit apparently caused aligment errors on ARM during the nightly test when initializing the 'Hw::Acpi_rsdp' structure.

skalk, 2019-05-16:
@chelmuth commit series f5dc456 9948eac and 2dc43ab solve this issue.

commit 50929a6e (2019-05-08): base-hw: fix compile errors with GCC 8.3.0

Fixes #3326
commit 70e13a3a (2019-05-16): base-hw: fix compile errors with GCC 8.3.0

Fixes #3326
closed 2019-05-29 by chelmuth